nodemcu/nodemcu-firmware #1885: Bug using sonoff device

jinxmcg, 2017-03-27:
### Expected behavior
Triggering fast switches of relayPin on Sonoff wifi device should start and stop as fast as possible.

### Actual behavior
When plugged into the power line, after about a dozen of fast on/offs it skips one HIGH or LOW and then works again and the issue repeats itself after another dozen. Powered by USB works without issues. At first I thought is  a hardware issue, but I wrote the same code in Arduino and uploaded to device. It works without any problems.

### Test code
```Lua
Led=7           -- gpio 13
ledPin = 4      -- gpio 2 
relayPin = 6    -- gpio 12

gpio.mode(ledPin,gpio.OUTPUT)
gpio.mode(relayPin,gpio.OUTPUT)
gpio.mode(Led, gpio.OUTPUT)

gpio.write(relayPin,gpio.LOW)
gpio.write(ledPin,gpio.LOW)
gpio.write(Led, gpio.LOW)

intervalTimer = tmr.create()

intervalTimer:register(100, tmr.ALARM_AUTO, function (t)     
	if (gpio.read(relayPin) == gpio.LOW) then
	    gpio.write(relayPin, gpio.HIGH)
	    gpio.write(Led, gpio.LOW)
	else
	    gpio.write(relayPin, gpio.LOW)
	    gpio.write(Led, gpio.HIGH)
	end;
end);
intervalTimer:start()
```
### NodeMCU version
nodemcu-master-9-modules-2017-03-01-14-52-36-float.bin built with nodemcu-build.com
(enduser_setup, file, gpio, net, node, tmr, uart, websocket, wifi.) I can provide the bin file.

### Hardware
I used a standard Sonoff wifi device.

FrankX0, 2017-03-30:
Looking at your description I still expect it is hardware related: powered by USB it works fine, and it only happens with fast on/offs.
With fast on/offs I guess you mean switching the relay every 0.1 seconds. Correct?

Your code is reading the status of the output. Not sure about the IO-cell of the ESP8266, but I guess it is actually reading the hardware state, not the software state.
Can you change your application by using an intermediate variable, which is used to switch between the two states. Depending on the state you then switch the relay on or off. This will rule out if the switching of the relay influences the program execution.

jinxmcg, 2017-04-11:
I think you might be right that is an issue of gpio.read(relayPin) to add a delay while reading, because in the Arduino code I just switched on an off without reading the relayPin.

I just did not have the time to test the new code it to confirm that you are right. 

I think is best to close this, and I will open if after testing I will experience the same issue.

Thank you for your time!

closed 2017-04-11 by jinxmcg