nodemcu/nodemcu-firmware #1890: ds3231.setTime DisableOscillator isn't optional

Buxxter, 2017-03-30:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/ds3231/ds3231.EN.md#settime

> disableOscillator: (optional) 0-1

### Actual behavior
disableOscillator isn't optional:
```bash
ds3231.lua:100: attempt to index global 'bit' (a nil value)
stack traceback:
	ds3231.lua:100: in function 'setTime'
	stdin:1: in main chunk
```
### Test code
```Lua
require("ds3231")
ds3231.init(5, 6)
second, minute, hour, day, date, month, year = ds3231.getTime();
-- Update Current Date
ds3231.setTime(second, minute, hour, 4, 30, 3, 17)
-- Get current time
print(string.format("Time & Date: %s:%s:%s %s/%s/%s", hour, minute, second, date, month, year))
ds3231 = nil
package.loaded["ds3231"]=nil
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: false
	modules: cron,dht,file,gpio,i2c,mqtt,net,node,ow,rtctime,tmr,uart,wifi
 build 	built on: 2017-03-25 23:46
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware
NodeMCU development kit v1.0
DS3231 module

8<------------------------ END BUG REPORT -------------------------------------

Buxxter, 2017-03-30:
Firmware built without 'bit' module.
Should check it in setTime too

devsaurus, 2017-03-30:
From what I see in the code, `disableOscillator` is optional. The error you got is only related to the lack of the `bit` module in the first place. Please include it in your build and try again.
However, I can't do deeper checks since I don't have the hardware available.

Buxxter, 2017-03-31:
> The error you got is only related to the lack of the bit module in the first place.
Yes, it's the only reason. Firmware with 'bit' works fine. Closed.

PR #1891 (2017-03-31): ds3231: require bit module, doc formatting
Fixes #1890.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Mainly formatting fixes for the docs and a `require("bit")` statement in the module code.
`ds3231.init()` function removed to ease coexistence of multiple slaves on the same I2C b [... cut]
closed 2017-03-31 by Buxxter