nodemcu/nodemcu-firmware #1892: Deprecate or remove tmy.delay()

TerryE, 2017-03-31:
### Issue

Use of `tmr.delay` invariably results in an application implementation that invariably breaches Espressif timing guidelines.  Moreover its use within Lua for short delays is misleading due to the overheads of the RTS.  If the programmer needs delays this precise, then the application should be recoded as C library anyway.

### Justification
`tmr.delay` is a misleading fossil that needs killed off.

### Example uses in the current code base.

*  `lua_examples/onewire-ds18b20.lua`.  Another fossil that deserves to go.
*  `lua_examples/u8glib/u8g_rotation.lua`. The rotation test should be rewritten to use a repeating timer/
*  `lua_examples/irsend.lua`.  An example of how to push Lua coding to the limit.  I suspect that the timing is so critical here that this depends on an Integer build an QIO flash and migt not even work with the new larger Flash-based builds.   This sort of application should be written as a C library of use `gpio.serout()`.  I certainly don't see this as an example that we should be promoting to new developers.  Pehaps @dvv Vladimir has the right to comment on this PoV.

All of the following misuse the delay function for sensor back-off periods.  Our official examples should *not* be doing this.  The correct mechanism is to use an alarm.
*  `lua_modules/ds18b20/ds18b20.lua`.  I have already suggest that we replace this.  I will submit the PR is there is any interest.
*  `lua_modules/bh1750/bh1750.lua:`  A delay of 200 mSec needs rewriting as an alarm callback.
*  `lua_modules/lm92/README.md`.  A delay of 1 sec needs rewriting as an alarm callback.
*  `lua_modules/hdc1000/HDC1000.lua  Sensor delays of 20 mSec need rewriting as an alarm callback.
*  `lua_modules/si7021/si7021.lua`. Ditto

vsky279, 2017-03-31:
Terry, we don't have anymore `tmr.delay` in `lua_modules/ds18b20/ds18b20.lua` and `lua_examples/onewire-ds18b20.lua` is gone.
Please see the PR https://github.com/nodemcu/nodemcu-firmware/pull/1820.

TerryE, 2017-03-31:
@vsky279 will comment on this PR re DS18B20 :smiley:

v1993, 2017-04-02:
I'm not sure that it's a good idea, we probably have to give everyone an opportunity to make non-official lua modules for protocols. I didn't udnerstood nodemcus' philosophy at all for now, it's just mine opinion.

TerryE, 2019-06-18:
One for me still to review.

TerryE, 2019-07-26:
On reflectiion we can't remove this because there are very valid uses of `tmr.delay()` as the ROM routine that this calls does allow ISRs to interrupt the delay, but accommodates this in the delay logic.  The current FAQ already deprecates its use at an application level.

closed 2019-07-26 by TerryE