nodemcu/nodemcu-firmware #1902: Not 100% work for the net.TCP server

METASHONGKONG, 2017-04-05:
### Problem
Firstly, Not 100% work for the net.TCP server. if I put the "conn:send("hello world")" in the receiver2 function, it is not 100% guaranteed that It works properly. 

It works fine for most of the testing. The below 2 are examples to illustrate the problem.

Example 1:
One time, I have tried the below code for one of the nodemcu dev kit 1.0, it always shows "no internet connection" rather then returning string hello world although I have rebooted it many times. However, if I put"conn:send("hello world")" in the listen 80 function directly (below code part I have commented), it works back for putting the "conn:send("hello world")" in the receiver2 function . Are there any caches?

Example 2:
After the nodemcu dev kit is opened, I type 192.168.4.1 in browser after the pc is connected the wifi emitted from the board. For the first trial, it cannot return the string. However, it works after around 30s-60s.

### Expected behavior
Not 100% work for the net.TCP server

### Actual behavior
Sometime, the server works properly, it return the string to the peer. However, it shows "no internet connection" rather then returning string hello world for sometime.

### Test code
```Lua
-- AP SSID/PW config               
apcfg = {}
apcfg.ssid = "Metas"..node.chipid()                    
apcfg.ssid = string.sub(apcfg.ssid,1,string.len(apcfg.ssid)-1)
apcfg.pwd = "12345678"

-- Wifi AP mode
wifi.setmode(wifi.SOFTAP)
wifi.ap.config(apcfg)  
print("AP mode started: "..apcfg.ssid.." "..apcfg.pwd)

function receiver2(conn,request)
	print("Received");
	print(request)	
	conn:send("hello world")
end

-- Create TCP server port 80
srv = nil
srv=net.createServer(net.TCP) 
srv:listen(80,function(conn) --pass to caller function as param if a connection is created successfully 

	--Client connected debug
	port, ip = conn:getpeer()
	print("---------------------")
	print("Client connected: "..ip..":"..port)
	--conn:send("hello world")
	
	--Register callback event
	conn:on("receive", receiver2)
	conn:on("sent",function(conn) conn:close() end)

end)  
```
### NodeMCU version
NodeMCU 2.0.0

### Hardware
Nodemcu dev kit 1.0
https://github.com/nodemcu/nodemcu-devkit-v1.0

djphoenix, 2017-04-05:
1. Which browser you're use? Maybe "no internet connection" message is result of its internal check.
2. Where are HTTP headers for your "hello world" response? It should be (at least):
```lua
function receiver2(conn,request)
  print(request)	
  conn:send(
    "HTTP/1.0 200 OK\n" ..
    "Connection: close\n" ..
    "\n" ..
    "hello world"
  )
end
```

marcelstoer, 2017-04-05:
Make that `\r\n` instead of just `\n` as per [HTTP/1.1 i.e. RFC2616](https://tools.ietf.org/html/rfc2616#section-4).

TerryE, 2017-04-06:
THe first thing that is immediately obvious is that you are assuming that service calls such as `wifi.ap.config(()` arer synchronous.  They are not.  What this does is to schedule an SDK work task to configure AP mode.  This task doesn't start after your function exits; yet within the same function you also `net.createServer()` and this library code is executed synchronously before the AP has been set up.  The `createServer` call should only be executed after the AP has been successfully configured and started.  The easiest way to do this is to move this call into an ALARM_SINGLE `tmr` callback after 5 seconds, say.

djphoenix, 2017-04-06:
@TerryE not sure, but why do you think that creation of server (Transport layer) infers with wifi connection (Interface layer)? Since it's not bound to specific IP/interface (and so bound to anycast - `0.0.0.0`), it's OK to create it in any time frame.

TerryE, 2017-04-06:
Yuri, I will defer to you re the LwIP net implementation.  It's just that in the old days, I found that putting a task break between any wifi config and the createServer() avoided initialisation problems.  Have you any alternative explanation for the OP's issue?

djphoenix, 2017-04-06:
Yep, and I was posted it before :) I think there is issue with HTTP headers.

TerryE, 2017-04-06:
> I think there is issue with HTTP headers.

No, the OP uses a bare TCP protocol, so HTTP issues don't apply here.

djphoenix, 2017-04-06:
In OP's message I see:
> After the nodemcu dev kit is opened, I type 192.168.4.1 in browser

So...

TerryE, 2017-04-07:
OK, oops, I missed that this isn't so much an HTTP issue but that the OP doesn't get that *browsers* use HTTP as the default transport protocol!

marcelstoer, 2017-04-07:
Still haven't seen anything that suggests a bug in our firmware. So, as per #1010 I'm closing this.

This code, put into `init.lua`, works for me. It's true that it's best to send HTTP headers if you're testing with a browser.

closed 2017-04-07 by marcelstoer