nodemcu/nodemcu-firmware #1904: u8g -> u8g2

joysfera, 2017-04-05:
### Missing feature
Please consider replacing the u8g module with [u8g2](https://github.com/olikraus/u8g2)

### Justification
The u8g is good for MCU with less than 512 bytes of RAM only, IMHO. ESP8266 has 100 times more RAM. The u8g requires unfamiliar and slow paging - four calls just for updating even so small diplays like Nokia 5110 with just 84 bytes of video RAM. This is ridiculous. ESP8266 can dedicate few hundred bytes for a video RAM easily thus u8g2 is totally appropriate and is far more straightforward to use, IMHO.

### Workarounds
None.

devsaurus, 2017-04-20:
Switching to u8g2 library would be the proper evolution step, also since upstream development of u8g (almost) stopped.
Though I'm not planning to look into this in the near future. Any takers?

joysfera, 2017-04-20:
How much work is involved? Is there any HOWTO for making NodeMCU modules from existing Arduino libraries?

devsaurus, 2017-04-20:
> Is there any HOWTO for making NodeMCU modules from existing Arduino libraries?

None that I am aware of, but there's a lot of information in [Porting to new MCU platform](https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform).

A rough outline of the workpackages, not necessarily in correct order:
- check inclusion strategy - verbatim copy of the files (like I did for u8glib) vs. git submodule
- port [GPIO and delay callback](https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#the-uc-specific-gpio-and-delay-callback)
- port [SPI and I²C callbacks](https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#communication-callback-eg-u8x8_byte_hw_i2c)
- generate Lua bindings for all functions of [U8g2](https://github.com/olikraus/u8g2/wiki/u8g2reference) and [U8x8](https://github.com/olikraus/u8g2/wiki/u8x8reference)
- provision (relevant) display setup Lua functions, both for [U8g2](https://github.com/olikraus/u8g2/wiki/u8g2setupc) and [U8x8](https://github.com/olikraus/u8g2/wiki/u8x8setupc)
- establish `#defines` for cloud builder (for display-at-interface combos and fonts)
- exemplary testing of an I²C and an SPI display
- generate API doc
- provide example scripts

pjsg, 2017-05-27:
Is there any progress? I'm looking to add a new driver and it makes more sense to do it for u8g2....

devsaurus, 2017-05-27:
I'm currently checking the feasibility to have u8g2 integrated in both esp8266 and esp32. Should take me a few days to come up with a first short in the esp32 realm. Then copy&paste to esp8266.

Would vote to include the source tree as a git submodule for esp8266 as well.

pjsg, 2017-05-27:
I'm agnostic as to a submodule / copies. In the latter case, we should provide a script that does the copying. I did this for the spiffs code... This makes it fairly easy to update. The one difficult file is the configuration which may have to be edited by hand.

pjsg, 2017-05-27:
Also see https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/U8G2 for a port to ESP-IDF

devsaurus, 2017-06-05:
A prototype integration of u8g2 for the esp32 firmware is available at https://github.com/devsaurus/nodemcu-firmware/tree/dev-esp32-u8g2. It needs some polishing and I'll add docs during the next days (see [graphics_test.lua](https://github.com/devsaurus/nodemcu-firmware/blob/dev-esp32-u8g2/lua_examples/u8g2/graphics_test.lua) for basic usage).

Porting this back to the esp8266 firmware should just require a specific implementation of the [platform hal code](https://github.com/devsaurus/nodemcu-firmware/blob/dev-esp32-u8g2/components/platform/u8x8_nodemcu_hal.c). And a viable solution how to share code between the platforms.

pastukhov, 2017-06-06:
I can't compile
```
pastukhov@pastukhov-Latitude-E5450:~/build/nodemcu/nodemcu-firmware-devsaurus$ make
CC u8g2.o
/home/pastukhov/build/nodemcu/nodemcu-firmware-devsaurus/components/modules/./u8g2.c:7:18: fatal error: u8g2.h: No such file or directory
compilation terminated.
/home/pastukhov/esp/esp-idf/make/component_wrapper.mk:176: ошибка выполнения рецепта для цели «u8g2.o»
make[1]: *** [u8g2.o] Ошибка 1
/home/pastukhov/esp/esp-idf/make/project.mk:386: ошибка выполнения рецепта для цели «modules-build»
make: *** [modules-build] Ошибка 2
```

devsaurus, 2017-06-06:
The u8g2 code is included as a git submodule. You're probably missing a `git submodule init --recursive`.
That should pull the upstream release.

pastukhov, 2017-06-06:
Thanks
It is compiled with ``` git submodule update --init --recursive ```

pastukhov, 2017-06-07:
Working with SSD1306 SPI and u8g2.ssd1306_128x64_noname driver!

v1993, 2017-11-13:
Any progress?

devsaurus, 2017-11-18:
I'll have a look at porting the ESP32 integration over to the ESP8266 firmware. It's more a matter of available time than feasibility.

devsaurus, 2017-11-20:
There's a functional version at https://github.com/devsaurus/nodemcu-firmware/tree/u8g2_port
Need to review the docs again and re-check submodule within the build system.

joysfera, 2017-11-20:
I'm going to give it a try ASAP, thank you!

joysfera, 2017-11-20:
I have cloned your repository, the u8g2_port branch, ran docker command and got this error:

```
In file included from u8x8_nodemcu_hal.c:8:0:
u8x8_nodemcu_hal.h:5:18: fatal error: u8g2.h: No such file or directory
 #include "u8g2.h"
                  ^
compilation terminated.
```

"u8g2.h" is nowhere in the repository, AFAICS.

Any idea what I have done wrong? Thanks!

joysfera, 2017-11-20:
Hmm, malloc is missing? How could I fix that?
```
u8g2lib/.output/eagle/debug/lib/u8g2lib.a(u8x8_d_fbrle.o):(.text.u8x8_d_overlay+0x0): undefined reference to `malloc'
u8g2lib/.output/eagle/debug/lib/u8g2lib.a(u8x8_d_fbrle.o):(.text.u8x8_d_overlay+0x4): undefined reference to `free'
u8g2lib/.output/eagle/debug/lib/u8g2lib.a(u8x8_d_fbrle.o): In function `u8x8_d_fbrle':
/opt/nodemcu-firmware/app/u8g2lib/u8x8_d_fbrle.c:90: undefined reference to `malloc'
/opt/nodemcu-firmware/app/u8g2lib/u8x8_d_fbrle.c:143: undefined reference to `free'
collect2: error: ld returned 1 exit status
../Makefile:339: recipe for target '.output/eagle/debug/image/eagle.app.v6.out' failed
```

devsaurus, 2017-11-21:
> Hmm, malloc is missing?

Strange, didn't fail for me. But that needs to be `c_malloc`, included from c_stdlib.h.

joysfera, 2017-11-21:
what does that mean? Should I go ahead and replace ```malloc``` with ```c_malloc``` in u8g2lib?

devsaurus, 2017-11-21:
Yes, in app/u8g2lib/u8x8_d_fbrle.c replace malloc with c_malloc, free with c_free and stdlib.h with c_stdlib.h. That should do the trick.
I'll check tonight why I haven't seen it in my build environment.

joysfera, 2017-11-21:
That helped, thanks. Going to flash the firmware, test it a bit and report back.

devsaurus, 2017-11-21:
Thanks for your feedback, that slipped through...
It's also fixed in my branch now.

joysfera, 2017-11-21:
@devsaurus works great, thank you! This is a great addition to NodeMCU@ESP8266.

joysfera, 2017-11-21:
hmmm, drawStr() doesn't seem to be coping yet. Will have to double check if fonts are indeed included.

devsaurus, 2017-11-21:
Yes, u8g2 is much cleaner.
Thanks for test driving the port! I'll prepare a PR soon.

v1993, 2017-12-02:
I'm trying to test u8g2 module on esp32, I've enabled it in `make menuconfig`, but `require 'u8g2'` fails, and no global `u8g2` aviable. What is wrong?

devsaurus, 2018-07-30:
Closing, PR was merged into `dev`.

closed 2018-07-30 by devsaurus