nodemcu/nodemcu-firmware #1907: UART1 not usable from lua because NUM_UART is 1 in app/platform/cpu_esp8266.h

horsteff, 2017-04-12:
The ESP8266 has two uarts, but only UART0 is usable from lua. According to the code in uart module and driver files UART1 seems well supported, but using it from lua is currently impossible due to the check for supported uart ids in app/modules/uart.c (MOD_CHECK_ID(uart, id), checks id with NUM_UART which is defined as 1 in app/platform/cpu_sp8266.h).

I used the docker NodeMCU build to create a firmware image from master branch but with NUM_UART defined as 2 in app/platform/cpu_esp8266.h and used it successfully to transmit data from lua code via TXD1 on an ESP-12 module. Is there any reason not to define NUM_UART as 2?

### Expected behavior
uart.setup(1, ...
uart.write(1, ...
Should setup UART1 and write to TXD1

### Actual behavior
uart.setup(1, ...
uart.write(1, ...
Both raise "uart 1 does not exist".

### NodeMCU version
master branch

### Hardware
tested with ESP-12 module

dnc40085, 2017-04-15:
If UART 1 were exposed, it would be transmit only since U1_RXD is also SD_D1 which is one of the data lines to the internal flash.

I don't see a reason why it couldn't be used as a transmit only serial interface or a debug output.

SpComb, 2017-04-17:
:+1 for this

I've been running a patched 1.5.4.1 firmware with `#define NUM_UART 2` for a while now, using `uart.setup(1, ...)`for DMX output on the second UART: https://github.com/nodemcu/nodemcu-firmware/compare/master...SpComb:dmx-uart2#diff-90eef6db68568af326362418e2196fdeR14

AFAIK this just prevents using the second UART for debugging output from the ESP SDK?

devsaurus, 2017-04-24:
PR merged, thanks guys!

PR #1934 (2017-04-24): Expose UART 1
Fixes #1907 .

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/en/*`.

The ESP8266 has two hardware UARTs available, unfortunately the second UART is transmit only due to the fact that the receive pin is shared with one of the flash data lines (SD_D1 to be specific), the UART 1 [... cut]
closed 2017-04-24 by devsaurus