nodemcu/nodemcu-firmware #1908: ESP32 build and flash working fine, but no NodeMCU modules available.

hbouzas, 2017-04-13:
8<------------------------ BUG REPORT -----------------------------------------
I successfully built the ESP32 firmware pulling the latest version from GitHub as described here https://nodemcu.readthedocs.io/en/dev-esp32/en/build/, 
flashed it using the guidelines as described here https://nodemcu.readthedocs.io/en/dev-esp32/en/flash/. 
It boots normally and I get
 lua: cannot open init.lua
All good till now.
 But then when trying to test the Nodemcu modules, the basic ones and which were included using make menuconfig,  I get
 ‘attempt to index global xxxx ( a nil value) 
for example, when trying to access node.heap(), I get
 ‘attempt to index global node ( a nil value) 

### Hardware
ESP32  and ESP32 Wroom
Firmware built on Linux

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2017-04-13:
There were similar reports of incomplete builds in #1703. The only substantial outcome so far was a suspected incompatibility with `make` on OS X (see https://github.com/nodemcu/nodemcu-firmware/issues/1703#issuecomment-273564070). My local linux system (OpenSuse Leap) uses GNU make 4.0 without any issues.

hbouzas, 2017-04-13:
Thanks devsaurus
Cleaned up a bit and moved to GNU make 4.1, seems to work now.

closed 2017-04-13 by hbouzas

devsaurus, 2017-04-13:
Good to hear @hbouzas.

FYI I've just update our repo to latest IDF.

sabareesan007, 2017-08-10:
Hi,
while running 'make all', I am getting /nodemcu-firmware-esp32/components/modules/component.mk:12: *** GNU make version 4.0 or above required.  Stop.
Running Mac OS
Is there any solution to overcome this?
or I need to install Linux?

Thanks