nodemcu/nodemcu-firmware #1911: building with CLIENT_SSL_ENABLE causes reset before user_main is called

quirkz, 2017-04-16:
With 2.0 sdk and dev/master branches there does not seem like there is any way to get a working build with CLIENT_SSL_ENABLE defined.  Always resets after user_start_trampoline in user_main.c and before user_init() is called.   I assume this is in the SDK.  I'm just wondering if there is simply not enough memory for nodemcu and libssl in 2.0 sdk.  I'm not sure how to debug this further.

marcelstoer, 2017-04-16:
Besides [enabling SSL/TLS](https://nodemcu.readthedocs.io/en/latest/en/build/#tlsssl-support) you also need to include the TLS module.

closed 2017-04-16 by marcelstoer

quirkz, 2017-04-17:
Yes I'm aware, I can build a working 1.5 tag build.  I assume you mean the LUA_USE_MODULES_TLS define.  

Dev build with nothing enabled does not reboot but enabling only SSL and TLS and it reboots before user_main.  I assume it's some sort of memory problem.  

call_user_start_trampoline -> "call_user_start" @ iram1_0_seg+4() is sdk init?  

Call to user_init does not happen and it resets.   

Any idea?  Can you build a dev branch with SSL enabled that does call user_init?  I'm working with release defines on, -Os, and integer, no -g, just to drop the memory burden as much as possible.  Tried many different settings.

quirkz, 2017-04-20:
So it seems my problem was the nodemcu/tools/esptool.py did not work with a text size past 0x80000.  Using the espressif esptool and it was ok.