nodemcu/nodemcu-firmware #1915: HTTPS requests do not work

nickrttn, 2017-04-18:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

A successful response

### Actual behavior

The response I'm getting (including debug info) is:
```sh
client handshake start.
client handshake ok!
client handshake start.
client handshake failed
Error: invalid protocol message
HTTP client: Disconnected with error: -28
HTTP client: Connection timeout
```

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```Lua
-- I had a working WiFi connection here.
http.get('https://google.com',
	nil,
	function(status, data)
		print(status)
		print(data)
	end
)
```

### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

I used the cloud build service: This was built against the 1.5.4.1-final branch and includes the following modules: cjson, file, gpio, http, net, node, pwm, tmr, uart, websocket, wifi, ws2812, tls. 

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

NodeMCU Amica v2.0, no special hardware.

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2017-04-18:
This seems like a dup of #1707

closed 2017-04-18 by marcelstoer

nickrttn, 2017-04-19:
I am using 1.5.4.1 though. I saw that issue and it seemed to only be about `dev` and `master` branches.

marcelstoer, 2017-04-20:
I'm sorry Nick, I'm a fool.

First, I didn't pay attention to the fact that you already said it was 1.5.4.1. in the initial description.
Second, while this _is_ a dup of one of _my_ issues I pointed you into the wrong direction (due to my first mistake). 

If I'm right this time, keeping fingers crossed, then this has got nothing to do with HTTPS, looks like #1546. The response you get from google.com is simply too large. I would expect that an unencrypted GET request fails as well.

nickrttn, 2017-05-03:
I've since switched from using the `http` module to `tls` on the `master` branch, and all is well, even though parsing is a little more of a pain. Thanks for your help anyway.