nodemcu/nodemcu-firmware #1917: feature request: module to support MAX3421E (mini usb host shield)

Ognian, 2017-04-18:
### MAX3421E host usb support
mini usb host shields now cost ~ 6 USD and can be attached to ESP8266 via SPI
the creators of the arduino mini usb host have started ESP8266 integration (Arduino): https://github.com/felis/USB_Host_Shield_2.0/tree/ESP8266

### Justification
There are a lot of HID Devices like keyboard or barcode scanners who can be attached only via USB Host to a nodemcu, many other devices have actually only a serial over USB profile. So I would suggest to start with just the HID profile...

### Workarounds
Don't know of a workaround.

Thanks
Ognian

marcelstoer, 2017-04-18:
I can't think of a use case which requires to attach a HID device to an ESP8266. However, as the linked project shows I'm obviously not creative enough. Maybe not the right question but have you considered using an [Omega2 plus one of their docks](https://onion.io/product-category/docks/) instead?

Applying #1010 I vote to close this unless the OP can resource the development himself.

djphoenix, 2017-04-18:
@marcelstoer I vote for this thread because I have some usecases that requires USB host...)
Anyway this is interesting module, but who can do it? This is a question...

marcelstoer, 2017-04-18:
> I have some usecases that requires USB host

Can you share to help kick start my imagination?

djphoenix, 2017-04-19:
- as OP noted: barcode scanners. E.g. tool for read barcode and simple query server for some info, and so show something via screen or LEDs.
- my project: WiFi-midi bridge - some controllers (e.g. [MPD32](http://www.akaipro.com/product/mpd32)) have direct MIDI ports, but some haven't (e.g. my favourite [Launchpad](https://us.novationmusic.com/launch/launchpad)). If I can connect controller via USB (that uses only MIDI signals) - it'll be perfect :)

Ognian, 2017-04-19:
Thank you for the Omega2 link, actually I din't know that. Although the HW price is good it would be "another different HW and SW piece in my puzzle", therefore I would rather go with a raspie ... 

BUT my use case is really "simple" I want to connect a barcode scanner to the nodemcu and today's barcode scanners are USB running the HID protocol ... so actually it is just sending a short string via USB. Other devices who have USB HID are e.g. scales [etc](https://en.wikipedia.org/wiki/USB_human_interface_device_class#Other_devices)

I would suggest that someone who is more experienced lay down the foundation of this, since then it can be expanded easier by the community. What I have already found is the [MAX3421E Programming Guide](http://pdfserv.maximintegrated.com/en/an/AN3785.pdf) and some C code on github:
-  https://github.com/jbkim/USB_Host_Max3421 or
-  https://github.com/stefanlinden/MAX3421E (this "looks" good)

I can help with testing, but I don't feel experienced enough with the nodemcu to lay down the foundation...

tjareson, 2017-06-17:
Would be great to have such a module.
In my case I would like to connect a thermal printer to wifi which only has a usb connection (which becomes a serial /dev/ttyUSBx on the computer when connected). Unfortunately there is no option to get direct serial connection on the highly integrated printer board, so I'm stuck with USB and found the max3421 as well. 
Another use case would be a WLAN enabled RFID reader which are coming as compact usb stick like devices acting as a keyboard. 
I'm also happy to help with testing of printers, rfid readers and actually a barcode scanner as well

closed 2018-02-24 by marcelstoer