nodemcu/nodemcu-firmware #1922: Not support SPI slave

iFish85, 2017-04-20:
I can see there are some slave-related codes in app/driver/api.c, but why platform_spi_setup() hardcode spi_master_init() ?
Also the document shows that "spi.SLAVE - not supported currently"
Is there any known serious issues on spi slave implementation?

Thanks

djphoenix, 2017-04-20:
> Is there any known serious issues on spi slave implementation?

Yes - this is still undone :) And noone who need it and have time+skill to implement.

iFish85, 2017-04-20:
> Yes - this is still undone :) And noone who need it and have time+skill to implement.
Fortunately, we need this slave function on our production...
I find the esp8266 SDK (NONOS and RTOS) already include SPISlaveSendData/SPISlaveRecvData, if these functions can works well, Lua api also can be implemented, right?
Anyway, I will try to implement it.

Thanks.

closed 2017-06-05 by iFish85