nodemcu/nodemcu-firmware #1923: bmp280

johndoe71rus, 2017-04-20:
which module use for bmp280 better ?
bmp085 as compatible with bmp180 or bme280. 
for bme280 need remove part with humidity from example.

mickley, 2017-04-21:
The bmp085 module only works with the bmp180, NOT the bmp280 or bme280.  

For those two, use the bme280 module.  Since the bmp280 does not include a humidity sensor, it will just return nil for humidity.  

I think you are confusing your sensors a bit here, make sure you understand which is which.

closed 2017-04-21 by marcelstoer

johndoe71rus, 2017-04-22:
ok.
i use code
`local alt=217 -- altitude of the measurement place

bme280.init(5, 4)
local T, P, H, QNH, MHG = nil

T, P, H, QNH = bme280.read(alt)
-- altimeter function - calculate altitude based on current sea level pressure (QNH) and measure pressure
P = bme280.baro()
curAlt = bme280.altitude(P, QNH)
  print("bmp280 data")

if T<0 then
  print(string.format("temperature=-%d.%02d", -T/100, -T%100))
else
  print(string.format("temperature=%d.%02d", T/100, T%100))
end
MHG = P / 133,3
print(string.format("pressure=%d.%02d", MHG/10 , MHG%10))
`
The first measurement is always an error.

` > dofile("bmp280.lua")
bmp280.lua:9: wrong arg range
stack traceback:
	[C]: in function 'altitude'
	bmp280.lua:9: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk
> dofile("bmp280.lua")
bmp280 data
temperature=29.46
pressure=736.00
>`
Need initial initialization of variables?