nodemcu/nodemcu-firmware #1924: Cron Bug while Disabled schedules called again

michan18, 2017-04-21:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
first = cron.schedule("* * * * *", function(e)
  print("First Tick")
end)
second = cron.schedule("*/2 * * * *", function(e)
  print("second Tick")
end)
third = cron.schedule("*/3 * * * *", function(e)
  print("third Tick")
end)

first:unschedule()

first:schedule("* * * * *")

At each minute it should print First Tick
### Actual behavior
For Each Two and third Minutes I am getting second Tick and Third Tick but not the first tick
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
first = cron.schedule("* * * * *", function(e)
  print("First Tick")
end)
second = cron.schedule("*/2 * * * *", function(e)
  print("second Tick")
end)
third = cron.schedule("*/3 * * * *", function(e) 
  print("third Tick")
end)
```
#

michan18, 2017-04-26:
@djphoenix Did you had a look of the above mentioned problem????

I think the basic problem lies in this section

`  luaL_unref(L, LUA_REGISTRYINDEX, cronent_list[i]);
  memmove(cronent_list + i, cronent_list + i + 1, sizeof(int) * cronent_count - i - 1);
  cronent_count--;
 `

pjsg, 2017-06-18:
I'm not even sure how this ought to work. The implementation seems to require that the entry returned by cron.schedule is retained in a variable, otherwise it gets deleted by the gc. However, the documentation doesn't retain the reference:

```
cron.schedule("* * * * *", function(e)
  print("Every minute")
end)
```

My inclination would be to have disabled entries deleted by the garbage collector, but otherwise retained.

michan18, 2017-06-18:
@pjsg well
`cron.schedule("* * * * *", function(e)
  print("Every minute")
end)`

this code would be working perfect for one cron schedule , but if we schedule more than 3 cron schedule and try to unschedule the second one, this would corrupt the first and third as well, so that what I was asking here

marcelstoer, 2017-06-18:
> The implementation seems to require that the entry returned by cron.schedule is retained in a variable, otherwise it gets deleted by the gc.

If that's the case then this is "wrong" IMHO because it's not what one would realistically expect. The sample you posted is a fire-and-forget schedule and that is expected to work find I'd say.

> My inclination would be to have disabled entries deleted by the garbage collector

That's a tricky one. Is there a sensible use case where you'd disable a schedule only to to re-schedule it a few seconds/minutes/hours later? If the GC were to clear this one then calling `disable` would (implicitly) be the same as setting the schedule variable to `nil`. I'd find that a bit odd to be honest.

pjsg, 2017-06-18:
If the gc can't reach a disabled entry, then it can't be re-enabled -- so there is no risk to delete it. If it can be reached, then it is possible that it will be re-enabled. If the entry is enabled, then it shouldn't be deleted by the gc.

marcelstoer, 2017-06-18:
> If the gc can't reach a disabled entry, then it can't be re-enabled -- so there is no risk to delete it.

Fully agree, must have been a misunderstanding with my comment, sorry. I guess all I really wanted to say is two things:

```lua
cron.schedule("* * * * *", function(e)
  print("Every minute")
end)
```
This must never be GC'ed even though the entry isn't assigned to a variable  (in contrast to `ent = cron.schedule`). Since I won't ever be able to obtain a reference to the entry there's no chance I could later disable/unschedule it.

```lua
ent = cron.schedule("* * * * *", function(e)
  print("Tick")
end)
-- We don't need this anymore
ent:unschedule()
```

The comment in above the `unschedule` seems to imply that we treat this as dead entry. Hence, it may be GC'ed. Yet, [the docs](https://nodemcu.readthedocs.io/en/latest/en/modules/cron/#cronentryunschedule) also state that

> Disabled schedules may be enabled again

in which case it must not be GC'ed. If unscheduled entries _couldn't_ be rescheduled what would be the point of the `unschedule` function (might as well call `ent = nil`)?

PR #2153 (2017-10-30): Make the CRON module work reliably
Fixes #1924.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

There were two bugs -- the memmove moved the wrong number of bytes (which could have led to a crash) and the lua_touserdata calls were trying to get the pointer from the wrong stack location. It is (somewhat) of a miracle that it worked at all!
closed 2017-10-30 by marcelstoer