nodemcu/nodemcu-firmware #1925: tools/esptool.py fails to program images over a certain size

quirkz, 2017-04-21:
While trying to build a firmware with CLIENT_SSL_ENABLE the larger output image would fail to program when using the nodemcu esptool.py.   Anything over 500k will fail.  Maybe less.  

Using the espressif esptool.py is ok. 

Verify also fails.  

**Output of espresiff tool:**

esptool.py v2.0-beta2
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Verifying 0x7260 (29280) bytes @ 0x00000000 in flash against ./bin/0x00000.bin...
-- verify OK (digest matched)
Verifying 0x8d6b0 (579248) bytes @ 0x00010000 in flash against ./bin/0x10000.bin...
-- verify OK (digest matched)
Verifying 0x5a000 (368640) bytes @ 0x000a0000 in flash against ./bin/0xa0000-8mb.bin...
-- verify OK (digest matched)
Verifying 0x80 (128) bytes @ 0x003fc000 in flash against ./sdk/esp_iot_sdk_v2.0.0/bin/esp_init_data_default.bin...
-- verify OK (digest matched)
Hard resetting...

**Output of nodemcu esptool.py on with same images/flash:**

./tools/esptool.py --port /dev/ttyAMA0 verify_flash         0x0 ./bin/0x00000.bin         0x10000 ./bin/0x10000.bin         0xa0000 ./bin/0xa0000-8mb.bin         0x3fc000 ./sdk/esp_iot_sdk_v2.0.0/bin/esp_init_data_default.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Verifying 0x7260 (29280) bytes @ 0x00000000 in flash against ./bin/0x00000.bin...
-- verify FAILED (digest mismatch)
Verifying 0x8d6b0 (579248) bytes @ 0x00010000 in flash against ./bin/0x10000.bin...
-- verify OK (digest matched)
Verifying 0x5a000 (368640) bytes @ 0x000a0000 in flash against ./bin/0xa0000-8mb.bin...
-- verify FAILED (digest mismatch)
Verifying 0x80 (128) bytes @ 0x003fc000 in flash against ./sdk/esp_iot_sdk_v2.0.0/bin/esp_init_data_default.bin...
-- verify OK (digest matched)

marcelstoer, 2017-04-21:
As per https://nodemcu.readthedocs.io/en/latest/en/flash/#esptoolpy you should always use the latest (upstream) version and not relay on our outdated copy in the `/tools` folder. Ever since #595 we pretty much agree that we should delete it.

closed 2017-04-21 by marcelstoer

quirkz, 2017-04-22:
Ok, thanks.  Wasn't quite clear from docs.