nodemcu/nodemcu-firmware #1926: missing return value or an event in case net connection fails?

joysfera, 2017-04-21:
Either I don't understand something fundamental in NodeMCU or we are missing an API for failed net connection.

If you run the test code below when your wifi is not connected to an internet (imagine your router or modem lost connection) then you get no information about the connect() call that failed to connect to the given server. None of the event functions get called. It will just sit there waiting for a miracle to happen, eating about 420 bytes of heap. Now if you call it say every minute then you soon run out of memory, right?

I thought that by avoiding the ```http``` module and writing my own implementation of that using the ```net``` module I'll avoid the memory leak but apparently it didn't help much.

What am I missing please? Can this be fixed in my code or do we need an API change?

### Test code
```Lua
function C()
  local conn = net.createConnection(net.TCP, 0)
  conn:on("receive", function(sck, c) print("receive") end)
  conn:on("connection", function(sck, c) print("connection") end)
  conn:on("disconnection", function(sck, c) print("disconnection") end)
  conn:connect(80, "whatever")
  print(node.heap())
end
```
```
> C()
42768
> C()
42336
> C()
41936
> C()
41512
> C()
41144
> C()
40664
> C()
40280
> C()
39920
> C()
39552
```

### NodeMCU version
master 22e1adc4b06c931797539b986c85e229e5942a5f
### Hardware
Wemos D1 Mini

marcelstoer, 2017-04-21:
An interesting pattern was presented at https://github.com/nodemcu/nodemcu-firmware/issues/1804#issuecomment-280690422

joysfera, 2017-04-21:
I see, using net.dns.resolve() to check if internet works at all. That's an interesting workaround.
Still, shouldn't this be handled in the net.connect() as well? Say by adding another event next to "connection" and "disconnection"?

marcelstoer, 2017-04-21:
Yes, I agree. There are lots of reasons why the connection to the target server cannot be established.

HHHartmann, 2017-07-01:
But we might need another callback to be able to know that there was no connection.
What about "connectionfailed"
Or should we just call the "disconnection" handle.

This seems to be somewhat like issue #2007.

marcelstoer, 2017-07-01:
Thanks @HHHartmann for linking this to #2020 (and thus to your fix for that one). It is indeed the same use case that's broken here as well.

However, I believe we're talking about _two_, albeit related, issues.
- memory is not freed on connection failure
- currently there's no event you could register a callback for

The first issue could potentially be fixed w/o addressing the second. Memory would be freed but none of the three existing events would be fired.

> should we just call the "disconnection" handle

That would be illogical to me. You can't disconnect from a connection that wasn't established in the first place.
As for the name of the new event, it's difficult to come up w/ something that wouldn't break consistency because it already _is_ broken. `receive` is a verb, `connection` is a noun and `disconnection` is...what?
- `failed` - past tense and too generic, what is it that failed?
- `connectionfailed` - sort of a half-sentence in past tense, if we were to pursue this I'd opt for `connectionfailure` instead

How about adding a callback to `net.socket:connect(port, ip|domain)` and do without a new event?

HHHartmann, 2017-07-01:
@marcelstoer the similarity to #2007 indeed just is about the use case.

I don't like the Idea of adding a callback because it is a different way of reporting results than the rest of the interface uses. I could however imagine to change the whole interface to callback(s) some time in the future.

But for now how about `failconnection` or `failedconnection`? it would be lexically nearer to `reconnection` and `disconnection`. But I am really not sure about it.

Should we then have another issue either for the memory leak or for the callback?

marcelstoer, 2017-07-01:
I'd track both in this issue but I'll leave that up to @joysfera.

I see your point about adding a (of course optional) callback to `net.socket:connect()`. However, the reason why I still believe it would be elegant is two-fold:
- "proximity" to function call, it'd be first thing that'd happen in case of an error while connecting
- contrary to the existing callbacks it can only ever be invoked once

In any case I could live with both approaches.

joysfera, 2017-07-01:
The callback sounds great to me. The events are basically tracking state of the (established) connection while the callback would report connection error. Makes perfect sense to me. Also agree with those two reasons @marcelstoer mentions above.

HHHartmann, 2017-07-24:
Just checked the docs of node.js (which we claim to be similar to) and they have lots of callbacks like ours but also 'error', 'lookup' and 'timeout'. So we might consider to go that way too.

TerryE, 2017-07-26:
Try adding 
```Lua
function P() for kv, in pairs(debug.getregistry()) print (k,v) end; end
```
and doing `C();P()` and followed by `P()` after any disconnect messages.

What is happening is that you are piling up stuff in the Lua registry.  The user data references are the dangling sockets and the functions are your callbacks.  

It's the  `socket.__gc` method that clears the `on()` references in the registry and this is only fired if the socket object has no live references. (The common bug here is to include the parent socket in the callback function as an upvalue thus creating an implicit reference to the socket in he registry so the GC is never fired, but you aren't doing this here.)

When I do this, the disconnects fire as the receiving server disconnects the sockets and the ESP does an orderly clean up.  There isn't any memory leak but the registry reference allocator does create some dangling references.  This isn't leak a standard Lua feature. (See PiL 28.3 for more discussion of how this works.)

If you don't want to eat memory resources then don't open lots of sockets.  Simple really.  Open one and wait for a connect and process it then close or wait for the disconnect, which will fire then the server end (or the ESP connection logic) times out the connection.  

This isn't a bug in the firmware, so I recommend that we close this issue, in line with out guidelines.

joysfera, 2017-07-26:
@TerryE you say "the disconnects fire as the receiving server disconnects the sockets". What if there's no connection established in the first place, like I explained in the issue when I created it?

Maybe that you would suggest me to not open another socket and check instead if the previous socket is still around. Then the "conn" would have to a global variable and I would have to take care of it manually (instead of relying on GC to clean it up for me). Also, I would have to set it to "nil" somewhere - say at the end of "disconnection"?

Perhaps I could force the GC to clean up the previous connection by setting the global "conn" to nil before creating a new connection? Would that work and would it be clean?

``
function C()
  conn = nil

  conn = net.createConnection(net.TCP, 0)
  conn:on("receive", function(sck, c) print("receive") end)
  conn:on("connection", function(sck, c) print("connection") end)
  conn:on("disconnection", function(sck, c) print("disconnection") end)
  conn:connect(80, "whatever")
  print(node.heap())
end
``

This is getting kinda complicated. We'd need set of complete code patterns covering basic scenarios in NodeMCU documentation plus a list of "don't do's". Plain API explanation doesn't show the hidden traps.

TerryE, 2017-07-26:
@joysfera, Petr "disconnection" is a little of a misnomer. It is any form of err response on an open socket or following a connection attempt.  Setting `conn = nil` before the create does bugger all.  A connect will subsequently fire a "connection" event or a "disconnection" event.  You could defer your receive:
```Lua
conn:on("connection", function(sck, c) 
   sck:on("receive", function(sck, c) print("receive") end) -- note sck NOT conn here to avoid upvalue
   print("connection")
end)
```
The disconnection will also fire if you have a pending on("receive") and the far end closes the socket.

joysfera, 2017-07-26:
@TerryE the main point of this issue is that you'll get NO RESPONSE. Are you aware of that?

TerryE, 2017-07-27:
@joysfera, Petr when you start paying me, then you can get on my case for my missing some point in your previous comments.  :wink: In the meantime my contributions are pro-bono, and alternative is no help.

As to why `conn=nil; conn=<something else>`, have you read my [Developer FAQ](https://nodemcu.readthedocs.io/en/dev/en/lua-developer-faq/) and PiL as I suggest?  Lua does not use a reference counting-based GC, but instead is a triggered mark-and sweep system.  It walks the directed graph formed by all resources starting at the _G, the Lua stack and the registry.  Any resource that is referenced is kept and the others are destroyed / freed / returned.  Doing "v=nil; v=newv` and `v=newv` are effectively identical from a resource veiwpoint.

Because of delete cascades, it can take up to 3 sweeps to reclaim all unused resources.  If the resources are allocated using the Lua system (as is the case with the `net` module), it is any dangling references such as in the registry or closed upvalues of in-scope functions what will prevent the resources being reclaimed.  

As to your usecase, you claim "**you**'ll get know response"; what you should really state is "I get no response".  I feel it unrealistic for you to expect the maintainers to try to spend hours trying to recreate your setup given the loose description you give. 

My suggestion is that you have to adopt a different mindset when programming a resource-limited IoT device (in contrast to a PC environment for example).

-  Don't start a second connection attempt if the first is indeterminate.  I can't understand the rationale for doing this, as it is pointless.
-  In my experience TCP connections timeout unless they are alive, but I can understand that there are execution paths within the Espressif code that don't timeout correctly (a possibility here is the dns resolution code hanging, but again we'd need a more specfic test case to track this down).
-  If you want a backstop timeout then use a `tmr` alarm to close the socket.

TerryE, 2017-07-27:
OK, if I add a couple of `net.dns.setdnsserver()` for servers 0 and 1 to some dead IP, then I get the sort of hang that you describe.  The `net` module is hanging in the dns lookup.  However closing the socket recovers as expected.  Doing the multiple calls without closing the previous is what is causing the leak. This is a pure programming error on your part. You should expect to tie up resources if you do this.  What is happening is that you are reassigning `conn` from the userdata for the first socket attempt to the userdata for the second, etc. but the resources are still in the registry, but have been orphaned from the app.  If you manually use `debug.getregistry()[N]:close()` to trigger the destructor then the GC clears everything up.

So your bug, not the firmware.  Having DNS lookup timeout handling is a new feature request.

closed 2017-07-27 by TerryE

joysfera, 2017-07-27:
@TerryE , I understand your note about paying you for NodeMCU related things as I also lead two OSS projects and contribute to several others. I appreciate your replies (as all other replies) in this thread. However, please note that I didn't expect anyone "_spending hours trying to recreate my setup because of the loose description_", because **I provided complete, exact and reproducible source code that exhibit the issue immediately** (that is within less than one minute).

As to your play with words "_you'll get no response, I get no response_" - the point is that **everyone** using NodeMCU will get no response. I submitted this issue not because I have a problem with NodeMCU, but because I wanted to help make it better for everyone. I myself worked around this and other issues already but I thought that submitting an issue would help improving the NodeMCU for all users.

If you believe that callers of `net:connect()` should also fire a timer in parallel and watch the status somehow and then close it eventually if nothing happens - all this manually instead of fixing the NodeMCU API and adding such "watchdog" internally, then I am OK with that. It's just another little trap for the NodeMCU users that at least got documented here in this closed issue.

Last remark: you managed to let it hang by disabling DNS but the connection can hang in other ways as well (as others mentioned in previous comments). To reproduce simply replace `whatever` by a valid IP address and you (actually not you but me, right?) will see that with interrupted internet connection it hangs the same way. Feel free to call it my bug but changing this to a _DNS lookup timeout feature request_ might not be correct, after all.

marcelstoer, 2017-07-27:
Terry, I admit that I don't fully understand all the details you presented - still, thanks for sharing. Therefore (maybe), I don't understand why you closed this.

Don't you think it'd be useful to have an easy option to react to failures establishing a connection? If I'm not mistaken that was essentially Petr's question/request. He never claimed this be a firmware bug.

My proposal was this

```
conn:connect(80, "example.com", function()
  print("Failed to establish connction.")
  -- You could do conn:close() here but wouldn't it be odd to close a non-established connection?
end)
```

TerryE, 2017-07-27:
My reason for closing this is that the symptoms that the OP were report were as a result of his application bug, pure and simple.  If you open lots of sockets, then you get an apparent memory leakage. If you only open one at a time then you don't.  If you tidy up you dangling references in the registry, then the GC collects these resources.

There is a separate issue that the Espressif API doesn't seem to have a time-out on nslookup, but this merits its own separate feature request.  but this is really a new feature rather than a bug in our net code.

If you want to implement a hard-stop timeout, then you can do this with a timer alarm, e,g.
```Lua
conn:connect(80, "example.com")
local conn_timeout = tmr:create()
conn_timeout:alarm(30000, tmr.ALARM_SINGLE, function() -- etc
```
so why should anyone realistically expect one of our developers to spend a few weeks development work because they can't be bothered to add the extra line or two of application code?  I don't mind carrying on the discussion, but it really isn't a bug in our firmware so that's why I closed the issue.  Feal free to reopen this if you think my analysis incorrect.

Our net interface is a wrapper around the Espressif code.  The Espressif code delivers certain callbacks which our net module wraps and delivers as Lua callbacks.  The issue here is that the Espressif code doesn't time-out under certain circumstances but instead hangs waiting for a response.   The code is now open-sourced iin our source tree, so Petr is free to go through it himself and work out what the actual issue is.

The non-OS model is event driven.  We can't change that.  We have a situation here that under certain circumstances, the Espressif code hangs.  Not good.  Maybe we need to work out why, but this isn't going to be a simple task.  Any volunteers? 

IMO, the correct way to deal with a rare hang scenario in what is a well determined directed event chain is to have a timeout to clear down and restart the chain.  You certainly don't just ignore then hang and repeatedly reissue the request then raise a bug saying that you're using up all of the memory resources by doing this. 

As this dialogue, I've already spent a bit of time showing Petr some diagnostic techniques to analyse the issues, and suggested the strategy that I broadly use to make this a non-issue.  I am not sure whether he's bothered to read my FAQ or PiL.  

As to me at the moment, I am still up to my eyes in building a house.  I have recently just updated my FAQ which still remains largely a single-author work.  I am making steady but slow progress on the Lua 5.3 migration, and I am just about to open-source my own provisioning system as a PR and detailed example of how to do this type work with no resource leakage.

Maybe @djphoenix Yuri, might be interested in taking a look at this, but I don't think that it merits any more of my time.  Sorry.

TerryE, 2017-07-28:
@joysfera Petr, having slept on this, I apologise if I got a bit ratty.  The documentation and decent examples are a bit lacking, and that's one of the reasons that I maintain the FAQ, but as I commented above none of the other committers or active users seem to want ot offer equivalent material and worked examples.  Not sure why.  And from answering beginners Qs, one thing that is very apparent is that most developers don't bother to read what is published to help them, so I am not really sure that adding to the corpus would help anyway. 

If you rewrote the firs few lines of your function:
```Lua
conn = net.createConnection(net.TCP, 0)
if not connarray then connarray ={} end
connarray[#connarray+1] = conn
-- etc
```
Then it's a bit more obvious why calling C() repeatedly slowly used up memory.  From an applications viewpoint, how you manage these timeouts is very application specific, for example with my provisioning system, if the service isn't replying for any reason then the app just carries on on the assumption that there's no updates.  For some of my loggers which run on a fixed tick, some jsut drop the sample, others collect the data and send it all once they've got a response.

And in terms of the "router going away", are you aware of the `wifi.eventmonitor` which already provide this service.   But the problem with this dialog as I see it, is that this isn't really about a confirmed bug in the firmware, but more about providing help on your programming techniques, and this sort of discussion really belongs on SO or esp8266.com, not here, which is why I think that this issue should be closed.

joysfera, 2017-07-28:
@TerryE I've been thinking about this for a six more hours as well. I definitely do not want you to waste your precious time on this because 1) I also had been building a house thus I know the amount of time and work required very well, and 2) I believe that Lua 5.3 is now the single most important upgrade for NodeMCU (apart from ESP32 support which might even be kinda related to NodeMCU Lua 5.3 rewrite/upgrade) so I'd like you to have time to focus on that.

Back to my `conn:connect()` example/issue: I'm OK with keeping the current API (that is with kinda missing callback for a failed connection) but the NodeMCU definitely needs much more detailed documentation to cover all the various aspects that are not obvious not only to newcomers. Let me show you few examples that are not clear (to me, at least):

1) can the `conn` be reused for another connection? If so, when is it safe to re-use it? How do I find out that the previous connection finished if there is no "connection failed" callback? I find crucial for NodeMCU documentation to include your example with the timer that basically runs the timeout call to indicate a failed connection. If I were more proficient in English I'd submit a PR for that part of documentation already.

2) does the `createConnection()` hold some valuable resources or is it the `conn:connect()` that allocates those 2 kB of memory? If the `createConnection` is "cheap" (in terms of memory on heap) then it might be good practice to create it at the start of my application and keep in a global variable, then just re-use it throughout the whole program whenever I need it. If it, however, consumes most of those 2 kB then it is better to always "end" it whenever one is done with the connection. But then, how does one "ends" it? You mentioned calling `close` on it. Why? The current documentation does not explain anything about the life cycle of a "client" (that is created by `createConnection`). One could assume that Lua's GC takes care of that but this whole issue is about it being a false assumption in certain cases.

3) when is it safe and when is it required to call `conn:close()`? Does that call release all the resources?

etc. etc.

Mind you, I am not complaining that NodeMCU documentation is in a bad state - quite the contrary! The documentation is very nice and complete, but when one is trying to avoid memory leaks it seems to be crucial to actually understand these innerworkings. And that returns us to your suggestion in your previous reply - to read the source code of NodeMCU since it's open source and as we all know the source code is usually the only definitive documentation. Even if this is technically the correct answer I can tell you from my experience in other OSS projects (including those that I lead) that this is not the right answer in terms of getting more users interested in NodeMCU.

In the past two years I attended all major OSS conferences in the Czech Republic and held excited talks about programming ESP8266 and ESP32 in Lua using NodeMCU. I would hate to admit that I tried to bring more people to a NodeMCU project that basically requires each programmer to study the complete source code of NodeMCU to understand how to program ESP in Lua properly and safely. Perhaps you understand me why I'm trying to help improve the programmer's experience.

Let me add one more example of source of issues when programming with NodeMCU. In the last two years we (NodeMCU users) learnt that multiple consecutive calls to `net.socket:send()` fail to do what the caller expects it to do. The documentation now mentions that clearly and offers a clean solution (or a work-around) this "feature". However, just yesterday, while reading through the open issues in NodeMCU project on GitHub I learnt that there was basically the same issue with `mqtt.client:subscribe()`. However, this is not documented yet so it's a potential source of new issues and problems. "Someone" should again issue a PR with just two lines that would document it - that would fix it (at least for those who read the documentation).

Oh, TL;DR. If you happened to read it all then I have a small surprise for you: the "router lost connection" I mention in the original issue above cannot be caught by `wifi.eventmonitor` as you've suggested. Note that it wasn't "ESP client lost connection to router" but "router lost connection to the internet". In reality it means that if anything breaks on the way from the router to the destination node someone tries to `conn:connect()` to, simply if any of those many "hops" becomes inaccessible, then the `conn:connect()` hangs the way I described and you have no chance to find it out using `wifi.eventmonitor`.

TerryE, 2017-07-28:
Both we and Espressif have cleaned up some of the nasties to make the event sequences more well determined.  One of these was to explicit prevent multiple sends being stacked.  

The net library uses the Lua Registry to stash resources much as described in PiL.  IIRC you can have one TCP server and up to 5 simultaneous client socket open (subject to memory resources being available).  The data structure are maintain in userdata so that metatable binding can be used.   The Lua cb functions are also registered.   They are set and cleared by the `socket:on()` statement ( a nil value for the function just deletes the old one).  They are also cleared by the socket's __gc method, but you need the socket to be marked for gc for this to happen, but open sockets are still registered and hence won' be GCed, which is why you need to do the explicit close, even on errors and timeout.  

As to the SDK scheduler, it is none-preemptive (read my FAQ).  the current running Lua task has to return control back to the SDK before any cb will be delivered, and the library wrapper gets the Lua function just-in-time from the registry, so there are no issues about pending events using an old cb.

As far as Lua 5.3 goes the two biggest improvements are that numvals only take 4 bytes instead of the current 8, so you get the convenience of a float build with the RAM footprint of an integer one.  However the biggest improvement will bee the ability to save Lua modules to flash and to run them directly from there, so the RAM will only be needed for R/W data, and not code or constants

joysfera, 2017-07-28:
I've been talking about re-using `conn` that is not open (either because previous connection failed and hung or because the `conn:connect()` succeeded and finished already). Still focusing on the example of a function that tries to connect somewhere regularly (say each minute) to report something (and might hang and stuff starts piling up). I believe this is a very common scenario so it would be good to have a "solid" example implementation of it (in the documentation ideally).

I can either try reusing `conn` myself when I get to my ESP8266 to see if it doesn't crash, watch the heap usage etc. to decide whether it could work or dive in the source code and try to understand the implementation. Though this does not guarantee that the implementation doesn't change quietly in the future so it would be best to document it the documentation (hoping that if implementation changes the documentation would reflect it).

BTW, maybe that this whole issue is sort of pointless now that I've read in git history that a fix for http module was merged that ensures it always reports what happened, including a failed connection. So I (and others that did the same) could simply return from net module back to http one (that we abandoned earlier due to its memory leaks). Things keep changing in NodeMCU waters and what one learnt last month might no longer be valid today.