nodemcu/nodemcu-firmware #1928: ESP32 - Bluetooth controller not enabled

zelll, 2017-04-22:
### Expected behavior
BLE functions work.

### Actual behavior
BT does not work.
( BT works after adding ```esp_bt_controller_enable(ESP_BT_MODE_BTDM);``` in `lbthci_init()` in `bthci.c` )

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
bthci.scan.enable(1, function(err) print(err or "Ok!") end)
-- prints 31
bthci.adv.enable(1, function(err) print(err or "Ok!") end)
-- prints 31
```
### NodeMCU version
dev-esp32 587bfd4

### Hardware
ESP-32S module

devsaurus, 2017-04-22:
@jmattsson seems like the esp_bt API changed recently. I don't have a reference setup for testing the bthci module, can you check the updates please?

jmattsson, 2017-04-24:
Thanks! Both for the report and the fix :)
I've pushed an update for this (and updated both the IDF and the toolchain)

closed 2017-04-24 by jmattsson