nodemcu/nodemcu-firmware #1929: "xtensa-esp32-elf-gcc: cannot execute binary file" in compiling esp32 code in mac

helxsz, 2017-04-22:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

I am following the [instruction](https://nodemcu.readthedocs.io/en/dev-esp32/en/build/)  to download the source code from esp32 branch.
I have finished the step 'make menuconfig', however 'make' gave me a error saying 

'nodemcu-firmware-esp32/tools/toolchains/esp32/bin//xtensa-esp32-elf-gcc: cannot execute binary file
make[3]: *** [src/bootloader_flash.o] Error 126
make[2]: *** [bootloader_support-build] Error 2
make[1]: *** [/Users/aa/Documents/projects/nodemcu-firmware-esp32/build/bootloader/bootloader.bin] Error 2'

any idea what is problem with it, I know the instruction is based Linux, how would make it in MAC

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

https://github.com/nodemcu/nodemcu-firmware/tree/dev-esp32

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

esp32 based development board

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2017-04-22:
I'd guess that the prebuilt toolchain binaries which are included as a submodule in https://github.com/jmattsson/nodemcu-prebuilt-toolchains/tree/92842d3c5a3af36be66039e73d25286aa1d57929 are for Linux only.

Probably all you need to do for Max OS is to download Espressif's prebuilt toolchain as explained in http://esp-idf.readthedocs.io/en/latest/macos-setup.html and extract it to a directory outside of the repo. The path to the toolchain is set in https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/Makefile#L11, you'd need to adapt it to the location where you extracted.

Maybe @bwhitman can provide some details here, he reported successful builds on OS X in https://github.com/nodemcu/nodemcu-firmware/issues/1703#issuecomment-273564070.

helxsz, 2017-04-22:
@devsaurus Thanks. I followed your instruction it works. 

The procedure is simple enough.
1. remove or modify the name of old esp folder in the toolchain (tools/toolchain/esp32)
2. just download http://esp-idf.readthedocs.io/en/latest/macos-setup.html and extract it to the directory of (tools/toolchain/esp32)
3. make

closed 2017-04-22 by helxsz

darshil-doshi, 2018-04-30:
The following link http://esp-idf.readthedocs.io/en/latest/macos-setup.html is no longer available. Can someone help me fix the error?

marcelstoer, 2018-05-01:
Seriously, is it too much to ask to go to http://esp-idf.readthedocs.io/ and type "macos" into the search field?