nodemcu/nodemcu-firmware #1930: WPS IP Problem

startimeahmet, 2017-04-22:
### Expected behavior
After connecting with WPS an IP is assigned to the ESP.
### Actual behavior
I get SUCCESS! message on the terminal. WPS connection is successfull, wifi credentials are taken but wifi.sta.getip() returns nil.
### Test code
```Lua
wifi.setmode(wifi.STATION)
wps.enable()
wps.start(function(status)
  if status == wps.SUCCESS then
    print("SUCCESS!")
  elseif status == wps.FAILED then
    print("Failed")
  elseif status == wps.TIMEOUT then
    print("Timeout")
  elseif status == wps.WEP then
    print("WEP not supported")
  elseif status == wps.SCAN_ERR then
    print("WPS AP not found")
  else
    print(status)
  end
  wps.disable()
end)
```
### NodeMCU version
2.0.0-master_20170202

### Hardware
Nodemcu Lolin v3 board (ESP-12E).

### Workarounds
Restarting the module connects ESP to WPS device since credentials is stored in flash. But why we would need a restart for WPS to work?

dnc40085, 2017-04-23:
Have you tried running `wifi.sta.connect()` after `wps.SUCCESS` status is returned?

startimeahmet, 2017-04-23:
Thanks a lot @dnc40085. It works okay now. Even though, I remember trying this method, I must have done something wrong.

closed 2017-04-24 by marcelstoer