nodemcu/nodemcu-firmware #1932: can't run the init.lua in esp32

helxsz, 2017-04-22:
using the Esplorer to upload the init.lua to esp32 based board (from the first compiled bin on my MAC)

### Expected behavior

when pushing the reset button on the esp32 board, should show a line of string 'hello NodeMCU' in the 
Esplorer console

### Actual behavior

the text shown on the console in uploading the code to esp32
```
stdin:1: attempt to call global 'w' (a nil value)w([==[--mqttConnected = 0]==]);
stdin:1: attempt to call global 'w' (a nil value)file.close();
stdin:1: attempt to index global 'file' (a nil value)dofile("init.lua");
cannot open init.lua
```

the text shown on the console in pushing the reset button on the esp32 board

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4172
ho 0 tail 12 room 4
load:0x40078000,len:9264
load:0x40080000,len:252
entry 0x40080034
[0;32mI (47) boot: ESP-IDF v2.0-rc1-597-gf73c6f8 2nd stage bootloader[0m
[0;32mI (47) boot: compile time 22:01:20[0m
[0;32mI (47) boot: Enabling RNG early entropy source...[0m
[0;32mI (66) boot: SPI Speed      : 40MHz[0m
[0;32mI (79) boot: SPI Mode       : DIO[0m
[0;32mI (91) boot: SPI Flash Size : 4MB[0m
[0;32mI (103) boot: Partition Table:[0m
[0;32mI (115) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (137) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (161) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (184) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (207) boot:  3 spiffs           unknown          c2 00 00110000 002f0000[0m
[0;32mI (230) boot: End of partition table[0m
[0;32mI (243) boot: Disabling RNG early entropy source...[0m
[0;32mI (260) boot: Loading app partition at offset 00010000[0m
[0;32mI (1068) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) [0m
[0;32mI (1068) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x0b070 ( 45168) map[0m
[0;32mI (1085) boot: segment 2: paddr=0x0002b080 vaddr=0x3ffc0000 size=0x01ddc (  7644) load[0m
[0;32mI (1115) boot: segment 3: paddr=0x0002ce64 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (1139) boot: segment 4: paddr=0x0002d26c vaddr=0x40080400 size=0x17b98 ( 97176) load[0m
[0;32mI (1211) boot: segment 5: paddr=0x00044e0c vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (1212) boot: segment 6: paddr=0x00044e14 vaddr=0x00000000 size=0x0b1f4 ( 45556) [0m
[0;32mI (1228) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x49f20 (302880) map[0m
[0;32mI (1254) cpu_start: Pro cpu up.[0m
[0;32mI (1266) cpu_start: Starting app cpu, entry point is 0x40080dd8[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1299) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1321) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (1342) heap_alloc_caps: At 3FFC6630 len 000199D0 (102 KiB): DRAM[0m
[0;32mI (1363) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1384) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1406) heap_alloc_caps: At 40097F98 len 00008068 (32 KiB): IRAM[0m
[0;32mI (1427) cpu_start: Pro cpu start user code[0m
[0;32mI (1484) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (204) cpu_start: Starting scheduler on APP CPU.[0m

Mounting flash filesystem...
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
lua: cannot open init.lua
```

### Test code

```Lua
print("hello NodeMCU")
```
### NodeMCU version

latest version esp32 branch

### Hardware
esp32 supported development board

devsaurus, 2017-04-23:
> stdin:1: attempt to index global 'file' (a nil value)dofile("init.lua");

This indicates that the `file` module (and probably all other modules) wasn't linked properly into the firmware binary. We had similar reports for Mac OS and my best guess is you update make like described in https://github.com/nodemcu/nodemcu-firmware/issues/1703#issuecomment-273564070.

devsaurus, 2017-04-23:
I investigated this issue a bit deeper: make 3.8.2 fails to expand MODULE_NAMES with the uppercase function in [`components/modules/component.mk`](https://github.com/nodemcu/nodemcu-firmware/blob/9cca38767f56645d455e01d2690830195801f912/components/modules/component.mk#L15) while 4.0 succeeds here.

I've [added a check for the make version](https://github.com/nodemcu/nodemcu-firmware/commit/9cca38767f56645d455e01d2690830195801f912) which errors out if anything below 4.0 is used.

closed 2017-04-24 by marcelstoer