nodemcu/nodemcu-firmware #1936: What exactly does wifi.STA_GOTIP mean?

eyaleb, 2017-04-25:
I do not know that this is a bug, but doco clarification can help. I logged a query on http://www.esp8266.com/viewtopic.php?f=24&t=14204 and got no response. Here is what I said there.

I set a static IP, so there is no need to access the gateway. Furthermore I only send (from the esp) UDP packets which are allowed to be lost.

Nevertheless, if I disconnect the a/p from the server (my gw) then after a few minutes I only get to status 3. When I reconnect the server the status soon progresses to 5 (STA_GOTIP).

I expected that once I associate with the a/p I instantly progress to STA_GOTIP status.

**Why is this important?** My app reads a sensor and sends a packet in under 500ms (usually). As it is now, when the server is down, a long timeout is incurred (about 5 seconds) which drains the battery 10 times faster.

The timeout is required to deal with intermittent (rare) a/p connection failures, but is unacceptable as a regular event.

I could run something like this
```lua
wifi.sta.disconnect()
wifi.sta.setip({ip="192.168.2.30",netmask="255.255.255.0",gateway="192.168.2.7"})
wifi.sta.config("SSID", "PASS", 1)
wifi.setmode(wifi.STATION)
wifi.sta.autoconnect(1)
```
and then check that everything is set
```lua
print(wifi.sta.status(), wifi.sta.getip())
=wifi.sta.getconfig()
```
The status shows 3 when the a/p cannot reach the gateway.

BTW `wifi.eventmon(wifi.eventmon.STA_GOT_IP,...)`  does not trigger either.

dnc40085, 2017-04-26:
If the Station(ESP8266) can't find the configured Access Point(AP or gateway), the WiFi disconnect event, `wifi.eventmon.register(wifi.eventmon.STA_DISCONNECT, myfunc())`, should return reason NO_AP_FOUND(201).
 
As for the `STA_GOT_IP` status, although there is no documentation stating so, I believe it means that in addition to having a valid IP, the station(ESP8266) is ready to communicate with the network.

The following code should help you deal with the `NO_AP_FOUND` disconnect event, so you can minimise battery use.
```lua
wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, function(T)
 print("\n\tSTA - CONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
 T.BSSID.."\n\tChannel: "..T.channel.."\n") 
end)

wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, function(T) 
 print("\n\tSTA - GOT IP".."\n\tIP: "..T.IP.."\n\tMASK: "..
 T.netmask.."\n\tGW: "..T.gateway.."\n\tStation ready!\n")
end)

wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T) 
 print("\n\tSTA - DISCONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
 T.BSSID.."\n\treason: "..T.reason.."\n")
  if T.reason == wifi.eventmon.reason.NO_AP_FOUND then
    print("Access Point not found, disconnecting...")
    wifi.sta.disconnect()
  end
end)

wifi.setmode(wifi.STATION)
wifi.sta.disconnect()
wifi.sta.setip({ip="10.0.0.153", netmask="255.255.255.0", gateway="10.0.0.1"}) 
wifi.sta.config({ssid="test_ap", pwd="longpassword", auto=true})
```
If an IP address is assigned to the station using `wifi.sta.setip()` before connecting to AP, the event `wifi.eventmon.STA_GOT_IP` should occur immediately after the `wifi.eventmon.STA_CONNECTED` event.

eyaleb, 2017-04-26:
First, I always connect to the a/p. I only send UDP messages, so I do not need any check/confirmation that the gw is accessible (I expect the packet to be quietly lost in this case).

> If an IP address is assigned to the station using wifi.sta.setip() before connecting to AP, the event wifi.eventmon.STA_GOT_IP should occur immediately after the wifi.eventmon.STA_CONNECTED event.
> 
i agree with this, and I do `setip()` very early.
Normally I do get `STA_GOT_IP` triggered immediately after `STA_CONNECTED`. This is the reason for this report, as I find that when the gw is unreachable, this does not happen. Something is blocking this final transition. Should this happen? Can I avoid this?

closed 2019-06-08 by TerryE