nodemcu/nodemcu-firmware #1937: invalid spec (val 15 out of range 0..6) at cron.c

farcop, 2017-04-25:
### Expected behavior
```Lua
cron.schedule("* * 15 * *", function(em) 
)
```
must create monthly call of function(em) on every 15th day of mounth.

### Actual behavior
```Lua
cron.schedule("* * 15 * *", function(em)
) 
```
brings exception
invalid spec (val 15 out of range 0..6)

### Test code
```Lua
cron.schedule("* * 15 * *", function(em)
) 
```
### NodeMCU version
latest master

### Hardware
wemos d1 mini

I suppose that it is order operations mismatch at https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/cron.c at lines 70-75
```
  desc->dow = lcron_parsepart(L, s + 1, &s, 0, 6);
  if (*s != ' ') return luaL_error(L, "invalid spec (separator @%d)", s - str);
  desc->dom = lcron_parsepart(L, s + 1, &s, 1, 31);
  if (*s != ' ') return luaL_error(L, "invalid spec (separator @%d)", s - str);
  desc->mon = lcron_parsepart(L, s + 1, &s, 1, 12);
  if (*s != 0) return luaL_error(L, "invalid spec (trailing @%d)", s - str);
```
must be changed to
```
  desc->dom = lcron_parsepart(L, s + 1, &s, 1, 31);
  if (*s != ' ') return luaL_error(L, "invalid spec (separator @%d)", s - str);
  desc->mon = lcron_parsepart(L, s + 1, &s, 1, 12);
  if (*s != 0) return luaL_error(L, "invalid spec (trailing @%d)", s - str);
  desc->dow = lcron_parsepart(L, s + 1, &s, 0, 6);
  if (*s != ' ') return luaL_error(L, "invalid spec (separator @%d)", s - str);
```

djphoenix, 2017-04-25:
This is already fixed in dev: https://github.com/nodemcu/nodemcu-firmware/pull/1884

marcelstoer, 2017-04-25:
@farcop thanks for the complete description an your own code analysis

closed 2017-04-25 by marcelstoer