nodemcu/nodemcu-firmware #1940: wifi.setmode() is persistent by default while wifi.sta.config() is not

marcelstoer, 2017-04-26:
The documentation says that [wifi.setmode()](https://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#wifisetmode) be persistent by default while [wifi.sta.config()](https://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#wifistaconfig) be not. Is that true and if so why should they behave differently? I don't understand that inconsistency.

dnc40085, 2017-04-26:
The documentation is indeed accurate, the reason for the inconsistency is due to issue #1288 in which I expressed concern about the implementation of a few WiFi functions and their potential impact on the lifespan of the flash. 
The change came in PR #1497 where I added the option to accept a table for station configuration, since the new version did not have to maintain backward compatibility I changed the default behaviour to be non-persistent.

The reason why I did it has to do with the following reasons:
* The SDK uses the functions `system_param_save_with_protect` and `system_param_load` to store wifi parameters in 3 sectors (0-1=data, 2=flag) located near the end of the flash. 
* Every time any persistent wifi setting is changed one of the data sectors and the flag sector are updated. 
* The flash ([Winbond w25q16bv](https://media.digikey.com/pdf/Data%20Sheets/Winbond%20PDFs/W25Q16BV.pdf)) has a erase/write limit somewhere north of 100,000 cycles

Example:
Lets assume that a project requires a WiFi parameter to be changed periodically(every ~2 min) and the developer mistakenly forgets to set the parameter to be non persistent in their script and puts the device into service unattended for an extended time. After ~5 months the device will have worn out the configuration sectors and will start producing errors.

Taking this info into account, I believe that it would be prudent to change the default behaviour for all relevant functions to be non persistent in order to extend the flash lifespan of any potential NodeMCU users.

marcelstoer, 2017-04-26:
Thanks a lot for digging out those references. Reading the comments there leaves me even more confused.

In #1288 both @jmattsson and I said that we wanted to keep the (then) existing defaults. Yet, when that big PR was up for discussion we didn't insist. I don't know about Johny but I simply didn't notice the relevant pieces in those 100s of modified LoCs.
In #1288 you also established that the SDK "doesn't overwrite the credentials if they are the same". Hence, it that case there's no wear on the flash. That to me was a strong argument for persistent settings by default.

> Lets assume that a project requires a WiFi parameter to be changed periodically(every ~2 min) and the developer mistakenly forgets to set the parameter to be non persistent in their script and puts the device into service unattended for an extended time.

I learned around this community that I'm really terrible with thinking up creative IoT use cases but still: how realistic of an example is that? And how likely is it that _both_ conditions are met (i.e. such frequent change _and_ wrong parameter)?
In my own software I always strive to optimize for the regular cases rather than for the corner cases. I would argue that the regular case be "more or less static WiFi settings"? Of course I'm eager to learn about all the other scenarios which would prove me wrong.

Allow a "counter" example:
EUS is one of our top-20 modules (according to my build stats). If you use it to configure your WiFi / AP connection you delegate calling `wifi.setmode()` and `wifi.sta.config()` to that module. I would assume that if you use that module you want your WiFi settings to be persistent - it'd be rather inconvenient to use otherwise. Is that case covered?

I've learned to appreciate your clear thinking and therefore I ask myself: where's my logic flawed?

dnc40085, 2017-05-02:
>I learned around this community that I'm really terrible with thinking up creative IoT use cases but still: how realistic of an example is that?

I agree, it's not a very good example as is, I was actually basing the example off an idea that I had for a wifi based keyless entry system. 
In this system, with the goal of preventing unauthorized access, the wifi parameters (MAC address, SSID and password) would be configured with values derived from a [TOTP algorithm](https://en.wikipedia.org/wiki/Time-based_One-time_Password_Algorithm) and would be changed every minute or two. 

>EUS is one of our top-20 modules (according to my build stats). If you use it to configure your WiFi / AP connection you delegate calling wifi.setmode() and wifi.sta.config() (or the C equivalent thereof) to that module. I would assume that if you use that module you want your WiFi settings to be persistent - it'd be rather inconvenient to use otherwise. Is that case covered?

With the EUS module, which I completely forgot about, I agree that the resulting station configuration should be persistent. However, I believe that there should be a notice in the documentation indicating that the EUS module modifies the persistent station configuration and that in doing so the on-board flash, which has a limited number of erase/write cycles, will be modified. 

I would say it's safe to assume that experienced developers would be familiar with the limitations of flash memory, but I would not make the same assumption with new developers and that is why I'm leaning towards non-persistence by default with the wifi module.

marcelstoer, 2017-05-05:
I agree with a lot of what you're saying, it makes sense but I'm not convinced yet. This might boil down to a trade-off between convenience (automatically persisted config) and damage prevention by the system (flash wear). We both tried to assess the likelihood of real damage done to flash memory and came to different conclusions.
I can live with either solution but it should just be consistent. Would be interested to find out where my fellow team members stand on this.

> there should be a notice in the documentation indicating that the EUS module modifies the persistent station configuration

Have you confirmed it actually does (I haven't)?

dnc40085, 2017-05-06:
>Have you confirmed it actually does (I haven't)?

Yes, the current implementation of the EUS module uses the persistent version (`wifi_station_set_config()`) of the station config function, as opposed to the non-persistent version (`wifi_station_set_config_current()`).
The station configuration is performed by `do_station_cfg` on line [623](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/enduser_setup.c#L623) of `app/modules/enduser_setup.c`

PR #1998 (2017-07-08): Update wifi.sta.config to save configuration to flash by default 
Fixes #1940.
 
- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/en/*`.

This PR updates `wifi.sta.config()` to save configuration by default.
closed 2017-07-08 by marcelstoer