nodemcu/nodemcu-firmware #1941: tmr limitation

danjohn, 2017-04-26:
### Missing feature
*tmr.interval()* only accepts intervals of 0..6870947(ms), i.e. 0..<2h. Although this is an inherited SDK limitation, this is unappropriate for over-the-day tasks, and should be overcome by the lua API.

### Justification
Imagine, the rtc is syncronized daily at 0:00, and it is wanted to light an ad panel from sunrise to sundip (so no fixed cron times are applicable); this would be best implemented using the more flexible *tmr* model.

### Workarounds
It can rudely be solved with a self-restarting (every ~2h) lua function. 
But extending the value range in the C API would save much eLUA (timer, var, func) ressources.

As the interval is obviousely kept in a 32bit var, it *should* be possible to use the full value range, resulting in intervals of ~50days (much more than sensible).
Something like
 ```
local delta=0
function tmr.alarm(id, time, mode, f) {
    if time<6870947 then existent:tmr.alarm(id, time, mode, f)
    else {
        delta=time
        tmr.alarm(id, delta, mode, function(di) {
            delta=delta-6870947
            tmr.alarm(di, delta, mode, @self)
        })
}
```
translated to the C layer would do the trick, not wasting the tiny ressources in the lua level and **not breaking the API** (except modifying the docs ;-)).

marcelstoer, 2017-04-26:
> no fixed cron times are applicable

There's no need for it to be fixed. You can schedule that 1-time interval, unschedule the entry when the time has come, do your job and the reschedule it for the next 1-time interval.

closed 2017-04-30 by marcelstoer

danjohn, 2017-05-30:
@m: Nix "schedule", it's about tmr, not cron:
Tmr isn't able to delay events over >2h, that's the problem!

I admit, there are technical reasons to restrict tmr in that way, but none logically.
And we "pro's" are urged to remove the technical ones in favour to the "users", I thought (and to keep consistent with "raw" LUA spcifications maybe) ;-)