nodemcu/nodemcu-firmware #1943: ESP32: WiFi powersave mode

FrankBuss, 2017-04-28:
I added "esp_wifi_set_ps(WIFI_PS_MODEM);" for test at the beginning of wifi_start in wifi.c for powersaving. It results in 20-30 mA power consumption (after a few initial seconds with 110 mA after connect), with some 110 mA spikes every few seconds. Without this line, the power consumption is all the time 110 mA.

Might be useful to add this a as a Lua function for the wifi-module, maybe just "wifi.setps" and as arguments wifi.PS_MODEM and wifi.PS_NONE. Looks like in newer versions of the framework there are more possible values, see http://esp32.info/docs/esp_idf/html/dc/df5/esp__wifi_8h.html

jmattsson, 2017-05-04:
Intended for inclusion under #1615, thanks for the measurements!

rvenkata, 2018-02-24:
hello
are there any examples of using the following modes?
WIFI_PS_NONE 	 No power save
WIFI_PS_MODEM 	 Modem power save
WIFI_PS_LIGHT 	 Light power save
WIFI_PS_MAC 	MAC power save

closed 2019-08-04 by stale[bot]