nodemcu/nodemcu-firmware #1944: ESP32: dsleep time limit and wrong documentation

FrankBuss, 2017-04-28:
The documentation in docs/en/modules/node.md is wrong for dsleep, it doesn't need pin32 and pin8 connected, at least I measured GPIO16 and reset (EN) of my ESP32 with a continuity tester and it was not connected, but nevertheless the dsleep command works. And I don't know how it is planned if it will be merged with the esp8266 code, if not, then the documentation should not note something for the esp8266.

The other problem is the maximum sleep time. The limit of 35 minutes in the current source code is because luaL_optinteger is used in node_dsleep in components/modules/node.c. This limits the number to positive integers, which is 2^31, so 35 minutes. If I change the first line to this:

```
uint64_t us = (uint64_t) luaL_optnumber (L, 1, 0);
```

then it works even for 2 hours. Might even be a problem for the esp8266 branch. I tested it like this:

```
frank@xps8900:~$ stty -F /dev/ttyUSB0 115200 -parenb -parodd -crtscts min 1 time 0 ; date ; echo -e "node.dsleep(7200000000)\n" > /dev/ttyUSB0 ; cat /dev/ttyUSB0 | ts       
Fri Apr 28 09:46:43 CEST 2017
Apr 28 11:43:02 ets Jun  8 2016 00:22:57
Apr 28 11:43:02 
Apr 28 11:43:02 rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

Not very accurate, but good enough. I guess it would work for much longer, too, because of the 64 bit value of the SDK, which is very useful, if you want to poll something, e.g. once per day. But this is for the float version of Lua, might need a comment in the docs that there is still the 35 minutes limit, if you use the int version of Lua.

jmattsson, 2017-05-04:
Ah, left-over from the ESP8266 branch. The ESP8266 does have the ~35min limitation, but we probably should allow longer sleeps on the ESP32 then. Tagging this as stuff to do as part of #1614, thanks!

tomsci, 2018-11-04:
> If I change the first line to \[use luaL_optnumber] then it works even for 2 hours

Only if your ROM isn't built with `LUA_NUMBER_INTEGRAL`. To support long timeouts even on 32-bit-integer-only builds I was wondering about making the API take a second optional `seconds` parameter, which if specified is multiplied up and added to the `us` parameter native-side.

That way you could write code like `node.dsleep(<bignumber_us>)` if you weren't bothered about only working on floating-point-supporting ROMs, but if you were, you could write `node.dsleep(0, <numsecs>)` instead. Or some combination if you really wanted a long but really precise sleep. Or we could just stop using microseconds for dsleep, as there's no real point to attempting a deep sleep at sub-second precision given the time taken to reboot (given we're not worrying about custom deep sleep wake stubs in NodeMCU, which is I assume the only reason the underlying API is in microseconds).

How bothered would people be to change the API to use seconds? Maybe make it a separate API, `node.deepsleeps()` to preserve compatibility?

It's not clear to me if the same max sleep time limit from the esp8266 also applies to the esp32. The [API docs](
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#_CPPv229esp_sleep_enable_timer_wakeup8uint64_t) usefully say "to be decided" and the TRM doesn't describe the limits for RTC_CNTL_SLP_TIMER1_REG. A comment in the code suggests an overflow calculating the RTC value might occur after 400 days but I don't know if there are other limits you'd hit sooner.

tomsci, 2018-11-05:
True, but that has the same problem if you disable floating-point support; suddenly the API is more restricted. I don't know how many people use integer-only Lua in their ROMs but I can't think of any other NodeMCU APIs that lose functionality if you do, except for `tmr.now()` and I [submitted](https://github.com/nodemcu/nodemcu-firmware/pull/2533) a PR to fix that :)

closed 2019-11-14 by stale[bot]