nodemcu/nodemcu-firmware #1946: mqtt module for esp32?

helxsz, 2017-04-28:
Just wonder when can we have mqtt feature for esp32? very looking forwards to this.

jmattsson, 2017-05-04:
As soon as someone capable finds the time for it.

The IDF now comes with the AWS IoT SDK which in turn includes an MQTT client, so in theory it's "just" about writing the Lua bindings for it. Of course, while at it we should probably also support the AWS IoT Device Shadow stuff. It's something I'd *like* to do, but I have had zero time to spend on ESP32 the last few months :(

vk2tds, 2017-07-10:
I am happy to be working on this, if someone is able to do a tiny bit of hand holding. My Git skills are a bit lacking and I would need some time to get up to speed. 

Having just looked at the AWS IoT SDK, it looks like this way would be the way to move forward. The issue with this strategy is that it is not compatible with the the ESP8266. 

AWS looks like it is a more supported way to move forward. One of the issues with the AWS code is that it does require certificates to operate, even within a local network. This is not a bad thing, but likely needs some documentation on how to set these up if you are doing a local install and not using AWS.

vk2tds, 2017-07-10:
I have done some further investigations. Following on from @jmattsson, it looks like a lot of the work has been done for us, and we need to build the Lua bindings to get things to work. This would mean that we no longer use the ESP8266 MQTT implementation, and the interface code is significantly edited. 

It looks like starting with this interface code and linking will be the way to go. Since we are dealing with protocols, it appears that much of the structure of the interfaces directly translates, which is good news.

One of the strange things looking at the code is that LWT (Last Will & Testament) appears to be named 'Will' in the Amazon/Espressif code. This is not a major issue.

Whilst the code is listed as being AWS, it does not appear that this is a requirement. Provided that the MQTT server is modern enough, and has TLS I doubt this will be a problem. I just found a quick tutorial on [Setting up TLS on Mosquitto MQTT](http://www.steves-internet-guide.com/mosquitto-tls/)

Further, the function aws_iot_mqtt_init() in aws_iot_mqtt_client.c forces certificates on the implementation. This might not be the desired situation when running your own MQTT server. The options are:
- Fork the Espressif code and make certificates optional, pushing changes back to the original
- Copy the Espressif code and modify it
- Force the ESP32 implementation to use certificates

My preference is for certificates. If NodeMCU on the ESP32 becomes really popular, and people build products on it, using certificates is likely a better way to do things anyway. 

AWS IoT Device Shadows look interesting, as they allow the AWS infrastructure to maintain details of the device state regardless of connection status. I haven't checked how hard they are to implement, but would require more work as there is no existing ESP8266 implementation. 

[Existing NodeMCU Interface Code](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/mqtt.c)

[Espressif Code](https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/7132505b00d2dd57f48478e75efa636021919aae/src)

jmattsson, 2017-07-10:
Indeed, while the aws-iot-sdk is for AWS, the core is still a standard MQTT client. Considering the history of issues in the ESP8266 MQTT client, I'm more than willing to give this client a go, even if it means a different Lua API for it. As a side note, here at work we're using the aws-iot-sdk in a production environment, albeit on different hardware.

Regarding the certificates, my suggestions would be to get a working implementation using certificates as a first port of call. Once that is done, and if there's a demand for it, looking at making it optional (by forking & submitting back upstream) could follow.

Device shadow support could be quite nifty and should map fairly well into Lua. The usual JSON compatibility quirk of a dedicated value for `null` would be needed, of course. There's another quirk with the device shadows themselves, where it is not natively possible to request the deletion of a value. Submitting a `null` in the `desired` branch just nukes the key/value there, and doesn't carry it through down to the client. The canonical way to deal with this is to use an empty array `[]` as the "please-delete-this-key-value". But, we're getting ahead of ourselves now :)

vk2tds, 2017-07-11:
I managed to get the ESP32-Dev to compile today, so my next step will be looking at getting the interface code to compile in and going from there. Work is pretty quiet for me at the moment, so the issue is more of ability rather than availability. 

Agree that we start with certificates and go from there - that was my thought too.

vk2tds, 2017-07-17:
An update... I have been working on this, getting my head around the code... And realised that I am not really the best person to be working on this, but I have been looking at it anyway.

Anyway, the issue I am having right now is code similar to the following [aws_iot_mqtt_client_publish.c](https://github.com/espressif/aws-iot-device-sdk-embedded-C/blob/master/src/aws_iot_mqtt_client_publish.c)

IoT_Error_t _aws_iot_mqtt_internal_publish():

```
/* Wait for ack if QoS1 */
  if(QOS1 == pParams->qos) {
    rc = aws_iot_mqtt_internal_wait_for_read(pClient, PUBACK, &timer);
    if(SUCCESS != rc) {
      FUNC_EXIT_RC(rc);
    }
  ...
  }
```

Essentially the code assumes that it is operating as a single process, not as a part of a bigger system. I suspect that going down this path may not be easy, without even more major works. Alas, I have a feeling we will need to go back to the ESP8266 implementation of MQTT and go from there. Ideally, this will require TLS being implemented as well, preferably up to version 1.2. 

Guess I have another rabbit hole to journey down... Thankfully all is not lost since I have a far better idea of how MQTT is implemented under NodeMCU now.

jmattsson, 2017-07-17:
I think it may be possible to run the entire MQTT client in its own RTOS task on the ESP32, and "just" post messages to & from it.

vk2tds, 2017-07-19:
Looking at the code ESP32, it appears that TLS is implemented already, which is good. I was looking at back porting the ESP8266 code across, and got stuck with the espconn references, but had another thought - implementing MQTT in Lua itself. 

A quick search has found the following MQTT Library [](https://github.com/geekscape/mqtt_lua/blob/master/lua/mqtt_library.lua), although it is implementing v3.1 rather than 3.1.1. The differences appear fairly minor [](https://github.com/mqtt/mqtt.github.io/wiki/Differences-between-3.1.0-and-3.1.1)

Given that the ESP32 has significantly more RAM (520 KB) and Flash (16MB), it might be easiest to implement MQTT in LUA. It can even be 'Compiled in' to the file system so it is always available.

Going through the code I also noted that HTTP is likewise apparently missing. It should be possible to implement this in a similar way

mdeneen, 2017-09-13:
I understand that the SDK has the amazon iot sdk.. but this looks somewhat useful:  https://github.com/tuanpmt/espmqtt

Thoughts?  It might be easier to get this integrated than the AWS stuff.  Especially if you want to make TLS optional, which would make migration from esp8266 easier.

devsaurus, 2018-10-02:
A PR to add `mqtt` module was merged.

closed 2018-10-02 by devsaurus

mdeneen, 2018-10-02:
https://github.com/nodemcu/nodemcu-firmware/pull/2501 for those interested

PR #2501 (2018-10-02): Mqtt rometatable
Fixes #1946.

- [x] This PR is for the dev branch rather than for master.
- [x]  This PR is compliant with the other contributing guidelines as well (if not, please describe why).
- [x]  I have thoroughly tested my contribution.
- [x]  The code changes are reflected in the documentation at docs/en/*.

> This pull request adds support for the mqtt protocol and is based on https://github.com/tuanpmt/espmqtt as suggested in issue #1946. This implementation simplifies mqtt support over the AWS IOT library implementation and has tried to maintain compatibility with the existing Nod [... cut]
PR #2307 (2018-10-02): Add MQTT module for esp32
Fixes #1946.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This pull request adds support for the mqtt protocol and is based on https://github.com/tuanpmt/espmqtt as suggested in issue #1946.  This implementation simplifies mqtt support over the AWS IOT l [... cut]