nodemcu/nodemcu-firmware #1948: ESP32: CAN module

zelll, 2017-04-29:
### Missing feature
ESP32 has a built-in CAN controller. We need a module for it in nodemcu ~

### Justification
Why not? :)

### Workarounds
MCP2515 + SPI ? 

---------
https://esp32.com/viewtopic.php?f=12&t=380 Discussion
https://github.com/eerimoq/simba/commit/46778af636a4fe58773753b5ed9da83916cfaa06 Driver
https://github.com/ThomasBarth/ESP32-CAN-Driver/ ESP-IDF based driver

zelll, 2017-04-29:
And here is a feature branch [dev-esp32-can](https://github.com/zelll/nodemcu-firmware/tree/dev-esp32-can/) ,based on @ThomasBarth 's driver.
Checked on an oscilloscope. I'll test it on a real CAN bus when I get my CAN device next week.

jmattsson, 2017-05-04:
Cool! I can't test it, but I'd certainly like to see it added when you've tested it!

PR #1958 (2017-05-07): ESP32: CAN module
Fixes #1948 .
- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

ESP32 has a built-in CAN controller. With a CAN transceiver (like MCP1551/TJA1050), ESP32 can communicate with other nodes on a CAN bus, for example, OBD on a car. This module can read/send standard frame [... cut]
closed 2017-05-17 by zelll