nodemcu/nodemcu-firmware #1952: feature request: report / read current gpio.mode(pin, mode [, pullup])

NiklausS, 2017-05-03:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.
I just run a couple of times into the case where i would have been happy to read the mode of a pin prior to set / change it. 
It's more a nice to have for verification than a really needed feature.
Thanks
 Nik

marcelstoer, 2017-05-05:
Are you aware of #1010? Are you willing to source the development yourself? Note that I didn't check whether the Espressif SDK even offers a function to query pin states.

NiklausS, 2017-05-12:
Thanks for the hint.

I've checked the SDK and the only thing in this direction i found is:
gpio_status = GPIO_REG_READ(GPIO_STATUS_ADDRESS);
As i'm working on a web-ide for the nodemcu i'm spending more time on this than on the firmware. 
If time allows i might dig deeper into this, for now i was quite happy to download the compiled firmware and concentrate fully to the lua.

Regards
Nik

marcelstoer, 2017-05-14:
> i'm working on a web-ide for the nodemcu

I'm sure you did some market research to figure out what the competition is doing but for the sake of completeness I'm including a link to my collection of tools: https://frightanic.com/iot/tools-ides-nodemcu/

> If time allows i might dig deeper into this

-> I'm closing this for now

closed 2017-05-14 by marcelstoer