nodemcu/nodemcu-firmware #1954: cron enhancement

danjohn, 2017-05-04:
### Expected behavior
Can feed cron with rtctime() timestamps

### Actual behavior
*cron.schedule()* only allows cron-like strings for the first argument. Using a timestamp like rtctime.get() gives, is quite odd.
''tm = rtctime.epoch2cal(rtctime.get())
tm["hour"] = tm["hour"] +5
cron.schedule(string.format("%d %d %d %d %d %d", tm["min"], tm["hour"], tm["day"], tm["mon"], tm["wday"], ), callback())''

### Justification
By now, it is very complicated to put (calculated) timestamps into cron. It's needed to break up time data into pieces using rtctime.epoch2cal(), reassembling that using string.format() and then reparsing that in the C layer (induces senseless overhead).

Making cron.schedule() aware of the signature (string cronPattern | int timestamp, [int repeat,] func callback()) would render out that shortcoming, and allow for calculated schedule times.

djphoenix, 2017-05-04:
You anyway need to format `repeat` value with something parseable. Now `cron` uses bitfield-masks for any part of cron-syntax schedule, so you can only repeat task by part values, not with any period that you want.
So I can agree for table-based syntax like:
```lua
cron:schedule({
  "*", -- all minutes - default string notation
  true, -- all hours - true means all possible values
  "*/5", -- each 5th days - default notation
  1, -- months - only january - one integer if single value
  true -- all wdays
}, cb)
```

marcelstoer, 2017-05-05:
As this is actually a side-effect of you having to overcome `tmr` limitations (I think... #1941) I don't support this proposal. It's corner case in my opinion. Also, I would argue that Yury's proposal adds unnecessary complexity which in turn means more chances for bugs.
I would stick to standard Cron as close as possible, [as much as it's standardized anyway](https://en.wikipedia.org/wiki/Cron#History). The reason for that being the "[principle of least astonishment](https://en.wikipedia.org/wiki/Principle_of_least_astonishment)."

If you externalize the Cron-expression building into its own function it isn't all that bad, is?

```lua
cron.schedule(cronForNowPlusFiveHours(), callback())

function cronForNowPlusFiveHours()
  tm = rtctime.epoch2cal(rtctime.get())
  -- doesn't consider things like day changeover, DST or leap seconds
  -- wouldn't you rather add 18000 seconds before calling epoch2cal?
  tm["hour"] = tm["hour"] + 5
  return string.format("%d %d %d %d %d %d", tm["min"], tm["hour"], tm["day"], tm["mon"], tm["wday"])
end
```

danjohn, 2017-05-10:
M, you guessed right with yr backref#1941 :)

I agree in not to blowing up the cron code with accepting to-be-defined tables (although someone might have use for that ITF, maybe).
@Yuri: I didn't had time yet to get familiar with the cron code, but I suppose it's not easy to parse timestamps/repeat intervals into the table?

@Marcel: The suggestion for a LUA-cron-expression-function is fine, but doesn't hit the problem:

Familiar use case: 
Set up the timing to manage the shadowing of a greenhouse; opening/closing light blinds for a day; or turning solar panels to get the right position in the morning.
All that has to be done at calculated sunrise/dip times from midnight (different each day).

Un/Rescheduling cron entries, not accepting timestamps (seconds from midnight), is no real option.
As tmr can't handle >2h, this also is no option by now.

I vote for working out #1941 in that case ;)!

marcelstoer, 2017-07-26:
As no committer has come forward actually supporting this proposal in the meantime I move to close it.

closed 2017-07-26 by marcelstoer

marcel, 2017-07-26:
FYI @danjohn every time you add @marcel to a message I get an email.