nodemcu/nodemcu-firmware #1957: Issue with integer management in runtime

landswipe, 2017-05-06:
### Expected behavior
If the 31st bit is set, it shouldn't cause the variable to become 0XFFFFFFFFF

### Actual behavior
See test code, just comment out either first or second line to feed value v differently. First line works fine without the 31st bit set, second line results in the value being 0xFFFFFFFF (incorrectly).

Interestingly the second invocation to bit.set(v,31) works fine indicating the problem is with the initial parsing of the value and not the internal checkInteger function itself. I have tested this with both bit and rfswitch (where I found it) modules, so it's not the module.

### Test code
```Lua
--v=0x7555AAAA  -- WORKS
v=0xF55FAAAA  -- FAILS and trashes value with 0xFFFFFFFF

v=bit.set(v,31)
print(string.format("%X\n", v))
v=bit.set(v,31) -- Second invocation to highlight context is fine, but initial parsing failed.
print(string.format("%X\n", v))
```
### NodeMCU version
master, 22e1adc4b06c931797539b986c85e229e5942a5f

### Hardware
NodeMCU, ESP8266, 4MB flash

FrankX0, 2017-05-07:
I can confirm this result, but only with a float build: with an integer build it works correctly.
In addition, it only fails when initially bit 31 is already set.
My guess is that this is caused by internal float representation and that your example can only work with an integer build.

landswipe, 2017-05-07:
Thanks for checking, I'm fairly new to the codebase and couldn't work out how to generate float vs integer images, can you let me know where it is configured?

FrankX0, 2017-05-07:
You can uncomment the following [line](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/include/user_config.h#L93) in `app/include/user_config.h`:
`#define LUA_NUMBER_INTEGRAL`

FrankX0, 2017-05-08:
Creating an integer build is now documented.
Please close this issue.

closed 2017-05-09 by djphoenix

landswipe, 2017-05-10:
Thanks Frank, so this behaviour is expected on float builds?

FrankX0, 2017-05-12:
Sorry, don't know.
I do not know how Lua is handling floats and integers in a float build.
Maybe @TerryE can answer this one?

TerryE, 2017-05-12:
The float version includes some "majic" in the `c_strtod()`, routine which is special to NOdeMCU builds.  If we change this then it might break other apps which assume this.  We are in the process of rebaselining NodeMCU on a Lua 5.3.4 baseline, but with 32 bit ints and floats (5.3.4 has separate number types for int and float.)   The main reason for this is speed: the ESP32 has H/W 32bit floating point, but 640bit is emulated in S/W.  We will introduce a userdata type for those that really want to have 64bit integer working.

PR #1961 (2017-05-08): Documenting creating integer build
Fixes #1957 .

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Documenting how to create an integer build.