nodemcu/nodemcu-firmware #1959: Code not returning correct temperature value.

aditgupta100, 2017-05-06:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Temperature data should be correct.

### Actual behavior
ds18b20 is giving temperature 34 while the expected temperature of the room is 25.

### Test code
```
    require('ds18b20') 
    ds18b20.setup(1) 
    temp_data=ds18b20.read()
    temp_data=ds18b20.read()
    print("Temp:"..temp_data.." C\n")
```

### NodeMCU version
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 22e1adc4b06c931797539b986c85e229e5942a5f

```

### Hardware
ESP8266 Nodemcu dev-board

8<------------------------ END BUG REPORT -------------------------------------

FrankX0, 2017-05-07:
Sorry, cannot reproduce your findings.
Because the FW only handles the OW protocol, your results mainly depend on the ds18b20.lua module.
Please check whether the issue is generated from this module.

FrankX0, 2017-05-12:
@aditgupta100: any progress on this issue?

marcelstoer, 2017-05-16:
No feedback from author in ten days -> closing

closed 2017-05-16 by marcelstoer