nodemcu/nodemcu-firmware #1960: ESP32: uart1/uart2 support

zelll, 2017-05-07:
I'm adding uart1/uart2 support to uart module. I noticed there had been some code for more uart(s) in `platform.c` and `platform.h` . So I tried to do it in the way expected...

I added `start()` and `stop()` functions, and changed signature of `on()` and `setup()`. Docs is updated [here](https://github.com/zelll/nodemcu-firmware/blob/dev-esp32-uart12/docs/en/modules/uart.md).
> uart.on(**[id],** method, [number/end_char], [function], [run_input])
> uart.setup(id, baud, databits, parity, stopbits, **echo_or_pins**)

@jmattsson Can you review it? 
https://github.com/zelll/nodemcu-firmware/tree/dev-esp32-uart12

jmattsson, 2017-05-08:
This is looking promising!

The docs seem to make the assumption that the console is always on UART 0. I don't know if that's necessarily true going forward?

I dropped a few comments in the branch you linked. Mostly small things. The one more conceptual question is how this will play with the `/dev/uartX` files/streams the IDF provides through the VFS. I haven't looked at how that works under the hood, but we should do something sane with regard to it. Possibly implement these uarts on top of that interface, if the `uart_write_bytes()` doesn't already play nice with it? @devsaurus have you poked around in this area lately?

devsaurus, 2017-05-08:
My understanding was that the `uart` module operates on the UART hardware/driver combo and is agnostic of the /dev/uartx streams. This appears to work interestingly well for stdout so far (VFS and `uart` module write to the UART driver via `uart_tx_char()`). But if someone ever tries to read from stdin then both might step on each others toes (VFS code directly accesses the UART HW regs).

Settling our `uart` module on top of the VFS stream abstraction would not be an option since there are no means to configure the stream in terms of terminal settings (ioctl and friends).

jmattsson, 2017-05-08:
Cool, sounds like the current approach is the sane way then. Thanks for the input!

zelll, 2017-05-08:
Some new ideas.
1. `uart.start(1)` takes more than 5 KB memory. I want to let `start()` return 0 or error code (returns `void` now). 
2. Make `uart.write()` returning _wrote bytes_, so lua can control data flow. 

seems ok?

PR #1982 (2017-06-13): ESP32: uart 1/2 supports
Fixes #1960.

- [x] This PR is for the `dev-esp32` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Added non-console UARTs (uart1 & uart2 when console is uart0, or uart0 & uart2 when console is uart1) to `uart` module.

Docs: [uart.md](https://github.com/zelll/nodemcu-firmware/blob/dev-esp32-uart12/docs/en/modules [... cut]
closed 2017-07-10 by zelll