nodemcu/nodemcu-firmware #1963: write large file , esp auto reboot

etimes2008, 2017-05-09:
nodemcu-firmware-2.0.0-master_20170202

spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)

vol = file.mount("/SD0")
if not vol then
  print("retry mounting")
  vol = file.mount("/SD0")
  if not vol then
    error("mount failed")
  end
end

function wget(url, filename, position)
    local PartialContentLength = 1024*200
    local contentLength, currentLength = 0, position
    local preTime = 0
    local tryFileCount = 5
    local RequestCount = 0
    local gcodeFile = nil
    local vol = nil

    local createRequest = function(path, host, from)
        local req = "GET " .. path .. " HTTP/1.1\r\n"..
        "Host: " .. host .. "\r\n"..
        "Connection: keep-alive\r\n"..
        "Accept: */*\r\n"..
        "Range: bytes=" .. from .. "-" .. from + PartialContentLength - 1 .. "\r\n" ..
        "\r\n"
        return req
    end

    wgetPartialContent = function(path, host)
        local partialContentLength = 0
        local currentPartialContentLength = 0

        local isGetInfo = false

        local parseResponse = function(resp)
            -- print("resp=>"..resp.."<=")
            local headerCode = resp:match( "HTTP/[%d.]+ (%d+).*\r\n" )
            partialContentLength = tonumber(resp:match( "Content%-Length: (%d+)\r\n" ))
            local payload = resp:match( "\r\n\r\n(.*)" )
            if payload then
                isGetInfo = true
                currentPartialContentLength = #payload
                -- print("down=====>",currentPartialContentLength, partialContentLength)
                gcodeFile:seek("set", currentLength)
                if not gcodeFile:write(payload) then
                    print("parseResponse write err!")
                    for i=1, tryFileCount, 1 do
                        gcodeFile:close()
                        gcodeFile = nil
                        repeat
                            gcodeFile = file.open("/SD0/"..filename, "w")
                            print("open file again", gcodeFile)
                        until gcodeFile ~= nil
                        gcodeFile:seek("set", currentLength)
                        if gcodeFile:write(payload) then
                            break
                        else
                            print("parseResponse write err! try==>"..i)
                        end
                    end
                    print("write done")
                end
            end
        end

        local c = net.createConnection(net.TCP, 0)

        c:on("connection", function(cc)
            local req = createRequest(path, host, currentLength)
            cc:send(req)
        end )

        c:on( "receive", function(rc, payload)
            if not isGetInfo then
                parseResponse(payload)
            else
                currentPartialContentLength = currentPartialContentLength + #payload
                pos = gcodeFile:seek()
                if not gcodeFile:write(payload) then
                    print("write err!", pos)
                    for i=1, tryFileCount, 1 do
                        gcodeFile:close()
                        gcodeFile = nil
                        repeat
                            gcodeFile = file.open("/SD0/"..filename, "w")
                            print("open file again", gcodeFile)
                        until gcodeFile ~= nil
                        gcodeFile:seek("set", pos)
                        if pcall(function(c) gcodeFile:write(c) end, payload) then
                            break
                        else
                            print("write err! try==>"..i)
                        end
                    end
                    print("write done!")
                end

                if #payload ~= 1440 then
                    RequestCount = RequestCount+1
                    if currentPartialContentLength >= partialContentLength then
                            currentLength = currentLength + partialContentLength
                            if (tmr.time()-preTime) > 0 then
                                print("download done!", currentLength, contentLength, currentLength/1024/(tmr.time()-preTime) .."KB/S", collectgarbage("count")*1024,RequestCount)
                            end
                            if currentLength < contentLength then
--                                gcodeFile:flush()
                                if RequestCount < 90 then
                                    isGetInfo = false
                                    rc:send(createRequest(path, host, currentLength))
                                else
                                    RequestCount = 0
                                    wgetPartialContent(path, host)
                                    c:close()
                                    c = nil
                                end
                            else
                                print("is download file done!", (tmr.time()-preTime))
                                RequestCount = 0
                                gcodeFile:close()
                                gcodeFile = nil
                                c:close()
                                c = nil
                            end
                    else
                            print("download fail!", RequestCount)
                            -- c:close()
                            if RequestCount < 90 then
                                isGetInfo = false
                                rc:send(createRequest(path, host, currentLength))
                            else
                                RequestCount = 0
                                wgetPartialContent(path, host)
                                c:close()
                                c = nil
                            end
                    end
                    -- print(collectgarbage("collect"))
                    collectgarbage()
                end
            end
        end )

        c:on("reconnection", function(cc, err)
            print("reconnection", err)
        end )

        c:on("disconnection", function(cc, err)
            print("disconnection", err)
            wgetPartialContent(path, host)
            cc = nil
        end )

        c:connect(80, host)
    end

    local proto, fullpath = url:match( "(%w+)://(.+)" )
    if not proto then
        proto = "http"
        fullpath = url
    end
    local host, path = fullpath:match( "([^/^:]+):?(.*)" )

    print("info=>", filename, host)
    
    wgetConn = net.createConnection(net.TCP, 0)

    wgetConn:on("connection", function(cc)
        print("connection")
        local req = "HEAD " .. path .. " HTTP/1.1\r\n"..
        "Host: " .. host .. "\r\n"..
        "Connection: keep-alive\r\n"..
        "Accept: */*\r\n"..
        "\r\n"
        cc:send(req)
    end )

    wgetConn:on("reconnection", function(cc, err)
        print("reconnection", err)
        currentLength = position
        wgetPartialContent(path, host)
    end )

    wgetConn:on("disconnection", function(cc, err)
        print("disconnection", err, currentLength, contentLength)
    end )

    wgetConn:on("receive", function(cc, payload)
        contentLength = tonumber(payload:match("Content%-Length: (%d+)\r\n"))
        print("contentLength", contentLength)
        preTime = tmr.time()
        if position == 0 then
            repeat
                gcodeFile = file.open("/SD0/"..filename, "w+")
                print("open file", gcodeFile)
            until gcodeFile ~= nil
        else
            repeat
                gcodeFile = file.open("/SD0/"..filename, "w")
                print("open file", gcodeFile)
            until gcodeFile ~= nil
            gcodeFile:seek("set", position)
        end
        currentLength = position
        wgetPartialContent(path, host)
        -- for i=0, contentLength, PartialContentLength do
        --     print("===>", i)
        --     wgetPartialContent(path, host, i)
        -- end
        wgetConn:close()
        wgetConn = nil
    end )

    wgetConn:connect(80, host)
end

wget("http://connect.iceman3d.net/new/2017-05/08/194081.stl.63374dbbd8dd00ef6c68c48cf4bceb90.gcode", "big.gcode", 0)
--wget("http://192.168.1.124/dragon.stl", "dragon.stl", 0)
--wget("http://connect.iceman3d.net/new/2017-05/02/195350.stl.2fd1b16f33a61e781ab2412ac0b74eda.gcode", "test.gcode", 0)

print:
parseResponse write err!
open file again	userdata: 3fff0148
parseResponse write err! try==>1
open file again	userdata: 3fff0148
parseResponse write err! try==>2
open file again	nil
open file again	userdata: 3fff0700

 ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x40100000, len 26844, room 16 
tail 12
chksum 0x5c
ho 0 tail 12 room 4
load 0x3ffe8000, len 2236, room 12 
tail 0
chksum 0x20
load 0x3ffe88bc, len 136, room 8 
tail 0
chksum 0x17
csum 0x17

etimes2008, 2017-05-09:
2g tf card is ok

closed 2017-05-09 by etimes2008

pastukhov, 2017-05-10:
Try to feed the watch dog with [tmr.wdclr()](https://nodemcu.readthedocs.io/en/dev/en/modules/tmr/#tmrwdclr)