nodemcu/nodemcu-firmware #1964: ESP32: GPIO 33 error

capella-ben, 2017-05-09:
### Expected behavior
I can config GPIO33 using:
`gpio.config({gpio=33, dir=gpio.IN, pull=gpio.PULL_DOWN})`

The documentation on GPIO indicates that ping 0 - 33 can be used.
[(https://nodemcu.readthedocs.io/en/dev-esp32/en/modules/gpio/#gpioconfig)]

### Actual behavior
When I send the gpio.config line it locks up towards the end.  
> gpio.config({gpio=33, dir=gpio.IN, pull=gpi

Reboot of ESP32 required.

Other pins (beside 33) seem to work fine including pins > 33.

### Test code
```Lua
gpio.config({gpio=33, dir=gpio.IN, pull=gpio.PULL_DOWN})
```

### NodeMCU version
using firmware build from 2017-05-08 version of dev-esp32

### Hardware
sparkfun esp32 thing

devsaurus, 2017-05-09:
Seems that we need to update the documentation to better reflect the available GPIOs in total. Along the lines of http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/gpio.html#overview.

Regarding your crash with GPIO33 - is it connected to anything specific on your module or externally?

devsaurus, 2017-05-09:
> Regarding your crash with GPIO33

Finally traced this down to a bug in the `gpio` module. Will prepare a PR to fix and also update the docs.

capella-ben, 2017-05-10:
Many thanks!

I tested this with and without something connected to the GPIO.

FrankX0, 2017-05-10:
According the [circuit diagram](https://cdn.sparkfun.com/assets/learn_tutorials/5/0/7/esp32-thing-schematic.pdf) I could find, GPIO33 (and 32) are connected to a 32kHz crystal. This could explain the reboot.

devsaurus, 2017-05-10:
Thanks for the hint, @FrankX0. I wasn't aware that there are modules that actually connect a xtal at these pins.
Mine doesn't but crashes as well, and here the crash is definitely caused by the way the module code talks to the gpio driver.

PR #1965 (2017-05-12): ESP32: Fix gpio pin mask generation and clarify available gpios.
Fixes #1964.

Generation of the pin_bit_mask did not work for indices >=32.
PR also adds an explanation which GPIOs are actually available. I intend to change the default statement in other module docs regarding pins with a reference to this section.

@capella-ben, @jmattsson please review for any additions or corrections.
closed 2017-05-12 by jmattsson