nodemcu/nodemcu-firmware #1971: ESP32: gpio.HIGH/LOW not defined

capella-ben, 2017-05-13:
### Expected behavior
When Running 

```
gpio.config({gpio=5, dir=gpio.OUT})
gpio.write(5, gpio.HIGH)

```
GPIO05 goes high (and onboard light is on)

### Actual behavior

```
> gpio.config({gpio=5, dir=gpio.OUT})
[0;32mI (377059) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0mgpio.write(5, gpio.HIGH)
stdin:1: bad argument #2 to 'write' (number expected, got nil)

```

If I replace gpio.HIGH with 1 then it works fine.  Seems like some constants aren't defined for me.

Could it be something that I have done wrong in the build?

Also note there is something wrong with new lines on output using ESPlorer.  I should raise a separate issue for this.

### Test code

```Lua
gpio.config({gpio=5, dir=gpio.OUT})
gpio.write(5, gpio.HIGH)
```
### NodeMCU version
dev32 branch current build.

### Hardware
Spark fun ESP32 thing

devsaurus, 2017-05-13:
Right, this is a bug in the docs as these constants are not available, thanks for pointing this out. Just use 1 or 0 instead of `gpio.HIGH` or `gpio.LOW`.
I plan to update gpio related docs anyway to have the pin index topic polished.

capella-ben, 2017-05-13:
Will do.  Thanks.  

Another thing I noticed is that for gpio.trig you mention INTR_UP, etc and it should be gpio.INTR_UP.

devsaurus, 2017-05-13:
I pushed a fix for the docs, they're in better shape now hopefully :)

capella-ben, 2017-05-13:
Many thanks!  Keep up the great work!

devsaurus, 2017-05-29:
Issues considered closed after update of the docs.

closed 2017-05-29 by devsaurus