nodemcu/nodemcu-firmware #1975: end user setup memory leak

Ser-Bor, 2017-05-16:
8<------------------------ BUG REPORT -----------------------------------------
Hi!
Sorry for my english.

May be I make something wrong? But in test code heap is decrease
### Expected behavior

### Actual behavior
Heapsize:38680
Heapsize:38416
Heapsize:38192
Heapsize:37848
Heapsize:37696
Heapsize:37072

### Test code
```Lua
tmr.alarm(1,10000,tmr.ALARM_AUTO,function()
  enduser_setup.start(
    function()
      collectgarbage()
      print("Heapsize:" .. node.heap())
    end,
    function(err, str)
      print("enduser_setup: Err #" .. err .. ": " .. str)
    end,
    print -- Lua print function can serve as the debug callback
  );
end)
```
### NodeMCU version
built on: 2017-04-18 08:46
Lua 5.1.4 on SDK 2.0.0(656edbf)
modules: enduser_setup,file,gpio,mqtt,net,node,ow,tmr,uart,wifi

### Hardware
ESP-07

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2017-05-16:
https://nodemcu.readthedocs.io/en/latest/en/modules/enduser-setup/#enduser_setupstart says

> Calling start() while EUS is already running is an error, and will result in `stop()` to be invoked to shut down EUS.

Did you test if the memory consumption is different if you manually invoke `stop()` with every iteration?

Ser-Bor, 2017-05-16:
> After an IP address has been successfully obtained, then this module will stop as if enduser_setup.stop() had been called.

> 
So i think that enduser_setup.stop() is not necessary here.
I can't test this with enduser_setup.stop() right now.

Ser-Bor, 2017-05-17:
Test it with enduser_setup.stop().
I have to add one more extra delay to start enduser_setup.stop() after enduser_setup.start() finished it work.
Leak still there:
### Actual behavior
Heapsize:38296
Heapsize:38216
Heapsize:38152
Heapsize:37960
Heapsize:37920
Heapsize:37848

### Test code
```Lua
tmr.alarm(1,15000,tmr.ALARM_AUTO,function()
  enduser_setup.start(
    function()
      collectgarbage()
      print("Heapsize:" .. node.heap())
    end,
    function(err, str)
      print("enduser_setup: Err #" .. err .. ": " .. str)
    end,
    print -- Lua print function can serve as the debug callback
  );
  tmr.alarm(2,10000,tmr.ALARM_SINGLE,function()  
    enduser_setup.stop()
  end)
end)

```

Ser-Bor, 2017-05-24:
Test it with firmware built on: 2017-05-24 10:30 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
Almost same result.

facat, 2017-07-28:
I also notice memory leak in my program. My firmware is powered by Lua 5.1.4 on SDK 2.1.0(116b762)
I think this firmware is not for production.

closed 2019-08-04 by stale[bot]