nodemcu/nodemcu-firmware #1977: [ESP32] Can't build cross.lua

feldim2425, 2017-05-17:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
That cross.lua compiles without errors

### Actual behavior
Error(s) while compiling cross.lua
```
:~/toolchain/esp32/nodemcu-firmware-32$ lua5.1 tools/cross-lua.lua 
[builder] Forcing rebuild due to configuration change.
[COMPILE] .build/cross-lua/app__lua__lapi.o
gcc: error: app/lua/lapi.c: No such file or directory
gcc: fatal error: no input files
compilation terminated.
[builder] Error building target
[builder] Last executed command was: 
gcc -O2 -Iapp/include -Iinclude -Iapp/lua -Wall -DLUA_CROSS_COMPILER -c -MD -MF .build/cross-lua/app__lua__lapi.d app/lua/lapi.c -o .build/cross-lua/app__lua__lapi.o
:~/toolchain/esp32/nodemcu-firmware-32$
```
### Test code
Just try to run ```lua5.1 tools/cross-lua.lua``` in the pulled dev-esp32 branch

8<------------------------ END BUG REPORT -------------------------------------

pastukhov, 2017-05-18:
Do you have lua headers installed?
```
liblua5.1-dev
```
For ubuntu.

feldim2425, 2017-05-18:
Yes "liblua5.1-dev" version "5.1.5-8ubuntu1" is installed.
I can also build the esp8266 version but not the esp32.
It seems like some files from the nodemcu-firmeware moved to different positions. For example there is no "app" directory in the branch for the esp-32

jmattsson, 2017-06-28:
There's an upcoming restructure of the repo that will merge the ESP8266 and ESP32 branches. This should be fixed then if not sooner.

devsaurus, 2019-05-31:
Duplicated by #2698.

closed 2019-05-31 by devsaurus