nodemcu/nodemcu-firmware #1981: ESP8266 ignoring global variable

hugalves, 2017-05-21:
Hi all,

I've been struggling myself to understand why is this happening, but I don't figured it out the problem. I have a simple script responsible for fetching google time. On ```receiver``` function, I print the time (this works) and set the time to a global variable ```now```. I don't know why, but this variable stills with empty value. 

Here it's the script:

```
now = ""

local function build_request()
  local payload = "HEAD / HTTP/1.1\r\n"..
                  "Host: google.com\r\n"..
                  "Accept: */*\r\n"..
                  "User-Agent: Mozilla/4.0 (compatible; esp8266 Lua;)"..
                  "\r\n\r\n"
  return payload
end

local function receiver(sck, callback)
  local time = string.sub(callback,string.find(callback,"Date: ")
         +6,string.find(callback,"Date: ")+35)
  print("receiver: "..time)
  now = time
end

function get_time()
  socket = net.createConnection(net.TCP, 0)
  socket:on("receive", receiver)
  socket:connect(80, 'google.com')
  socket:on("connection",function(sck)
    local post_request = build_request()
    sck:send(post_request)
  end)

end

get_time()
print(now)

```

Here are the responses:

```
-> nu terminal (nodemcu-uploader terminal shortcut)
--- Miniterm on /dev/cu.wchusbserial1410  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

> dofile('lib/bla.lua')

> receiver: Sun, 21 May 2017 15:35:57 GMT
```

Any help is really appreciated!

marcelstoer, 2017-05-21:
You got the answer at https://stackoverflow.com/questions/44081076/lua-ignoring-scope-variable and #1010 applies here as well -> closing

closed 2017-05-21 by marcelstoer

hugalves, 2017-05-21:
This code is an improvement of stackoverflow one, however, I got what I need. 
I still need help, why close?

TerryE, 2017-05-25:
> I still need help, why close?

Our [Support page](https://nodemcu.readthedocs.io/en/master/en/support/) explains our policy.