nodemcu/nodemcu-firmware #1983: ESP32: Error funtion of send with UDP server

LDmitryL, 2017-05-23:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
It is expected to echo received data

### Actual behavior
The send function in UDP server causes a system error:

`PANIC: unprotected error in call to Lua API (TestUDP.lua:15: bad argument #1 to 'send' (number expected, got string))`

But the data itself are taken!

### Test code

```Lua
serv=net.createServer(net.UDP) 
serv:on("receive",function(s,c) print(c) s:send("Echo:"..c) end) -- **ERROR ON THIS LINE**
serv:on("sent",function(s) print("echo donn") end) 
serv:listen(5999) 
```
### NodeMCU version
branch dev-esp32

### Hardware
module ESP-WROOM 32.
8<------------------------ END BUG REPORT -------------------------------------

djphoenix, 2017-05-23:
Was you see [note](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#netcreateserver) about `net.createServer` API for UDP sockets? And [`net.udpsocket`](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#netudpsocket-module) docs?

LDmitryL, 2017-05-23:
Of course not read. Relying on the sample programs. Thank you!

closed 2017-05-23 by LDmitryL