nodemcu/nodemcu-firmware #1984: MQTT failed to connect to the broker every couple of hours

Aterocana, 2017-05-23:
8<------------------------ BUG REPORT -----------------------------------------
The esp8266 is connected to a broker, another client publishes a message on topic ping every 10 seconds. When the esp8266 receives a message on topic ping, it publishes a message on topic pong.
### Expected behavior
The MQTT client remains connected to the broker.

### Actual behavior
Every 2 hours the MQTT client disconnects from the broker (reason -5). I'm losing 12 packages a day.
I've noticed every 2 hours starting from the startup (I have a timestamp which I didn't include here in order to provide a clearer code) the failing callback is triggered printing `Connection to broker failed, reason: -5`. It keeps exactly 2 hours from a disconnection to the following one.

### Test code
Here's the client I'm using:
```Lua
cfg = {
	ssid = "SSID",
	pwd = "PASSWORD
}
_id = "nodemcu2.1-" .. node.chipid()
_user = "USERNAME"
_password = "PASSWORD"
_broker = "BROKER_IP"

_send_topic = "pong"
_receive_topic ="ping"

_client = mqtt.Client(_id, 60, _user, _password, false)
_client:lwt("/lwt", _id .. " went offline", 0, 0)

_client:on("offline", function(client)
	print("MQTT: offline")
end)

_client:on("message", function(client, topic, data)
	_client:publish(_send_topic, "OK", 0, 0, function(client)
	end)
end)

connectMQTT = tmr.create()
connectMQTT:register(500, tmr.ALARM_SEMI, function(t)
	print("Connecting to the broker")
	_client:close()
	_client:connect(_broker, 1883, 0,
	function(client)
		print("Connected to broker")
		_client:subscribe(_receive_topic, _qos, function(conn)
			print("Subscribed to " .. _receive_topic)
		end)
	end
	,
	function(client, reason)
		print("Connection to broker failed, reason: " .. reason)
		connectMQTT:start()
	end)
end)
connectMQTT:start()
```
### NodeMCU version
I'm on NodeMCU 2.1 Master branch. I saw this problem also on previous version (2.0), while with 1.5.4.1 it seems working (it's up for a week now). I'm keeping ~10 devices with this code up 24/7.
The ones with firmwares 1.5.4.1 and 2.0 are up for more than a week now.
The ones with firmware 2.1 is up from yestarday (and they behave like the ones with 2.0 firmware). 

### Hardware
Regular esp8266 evaluation boards (with AI Thinker, WROOM02 and WirelessTag chips). 

8<------------------------ END BUG REPORT -------------------------------------

Let me know if you need any more informations since I'm able to set up many tests.

devsaurus, 2017-05-23:
Thanks for reporting this regression! Interesting to learn that the firmware is/was able to sustain such long periods at all 😉

However, even though I'm not very familiar with the mqtt code, I'd like to ask you to check whether the root cause for this issue can really be attributed to the mqtt module. Connection fail reason -5 translates to `MQTT_CONN_FAIL_SERVER_NOT_FOUND` which might indicate a problem with the underlying networking layer. Thus please check if you can correlate these drop-outs with any event from [`wifi.eventmon`](http://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifieventmonregister). Just to rule out issues from this direction.

Aterocana, 2017-05-24:
Ok, let me set up a couple of tests and in a couple of days I'll write down some results here. Do you have any clue about the different behaviour on 1.5.4.1 and 2.0 (or 2.1)? How come the first keeps longer while the latter fails every 2 hours? I've checked lua glue code and it seems nothing big happened (correct me if I'm wrong), so maybe the issue is due to a SDK change?
Moreover, in a 2.1 device I saw a lack of connection for a couple of minutes (where many packages were lost). I need more time to see if it's an isolated event.
Thanks for your job.

Aterocana, 2017-05-24:
I've a esp8266 running with the following Wifi Events handler:
```
wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, function(T)
	print("\n\tSTA - CONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
	T.BSSID.."\n\tChannel: "..T.channel)
end)

wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T)
	print("\n\tSTA - DISCONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
	T.BSSID.."\n\treason: "..T.reason)
end)

wifi.eventmon.register(wifi.eventmon.STA_AUTHMODE_CHANGE, function(T)
	print("\n\tSTA - AUTHMODE CHANGE".."\n\told_auth_mode: "..
	T.old_auth_mode.."\n\tnew_auth_mode: "..T.new_auth_mode)
end)

wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, function(T)
	print("\n\tSTA - GOT IP".."\n\tStation IP: "..T.IP.."\n\tSubnet mask: "..
	T.netmask.."\n\tGateway IP: "..T.gateway)
end)

wifi.eventmon.register(wifi.eventmon.STA_DHCP_TIMEOUT, function()
	print("\n\tSTA - DHCP TIMEOUT")
end)

wifi.eventmon.register(wifi.eventmon.AP_STACONNECTED, function(T)
	print("\n\tAP - STATION CONNECTED".."\n\tMAC: "..T.MAC.."\n\tAID: "..T.AID)
end)

wifi.eventmon.register(wifi.eventmon.AP_STADISCONNECTED, function(T)
	print("\n\tAP - STATION DISCONNECTED".."\n\tMAC: "..T.MAC.."\n\tAID: "..T.AID)
end)

wifi.eventmon.register(wifi.eventmon.AP_PROBEREQRECVED, function(T)
	print("\n\tAP - PROBE REQUEST RECEIVED".."\n\tMAC: ".. T.MAC.."\n\tRSSI: "..T.RSSI)
end)

wifi.eventmon.register(wifi.eventmon.WIFI_MODE_CHANGED, function(T)
	print("\n\tSTA - WIFI MODE CHANGED".."\n\told_mode: "..
	T.old_mode.."\n\tnew_mode: "..T.new_mode)
end)
```

I've got this:
```
Connection to broker failed, reason: -5
MQTT: offline
Connecting to broker
STA - GOT IP
Station IP: 192.168.0.152
Subnet Mask: 255.255.255.0
Gateway IP: 192.168.0.1
```

devsaurus, 2017-05-24:
Hm, are we looking at a DHCP lease renewal here? What is the lease time of your DHCP server?
Does the duration between drop-outs follow when you change the lease time?

Aterocana, 2017-05-25:
Nope, my DHCP lease lasts for 8 hours. And anyway, with SDK 1.5.4.1 this doesn't happen, just with 2.0 and 2.1.

devsaurus, 2017-05-25:
I can only speculate that the update of lwIP with #1435 also changed the DHCP client in this respect. IIRC the previous one did not renew leases after the initial IP offer - that would have been erratic behavior.

> Station IP: 192.168.0.152

Was the same IP assigned to this node during initial negotiation or a different one?
I'm asking because from my understanding there would be no need to down/up the interface if the IP didn't change. Just want to make sure we're not missing the root cause.

Aterocana, 2017-05-25:
It's the same IP. That was my dubt too, why putting down/up the interface if the IP is still the same?
I consider closed this issue, but I'm still seeing this problem:

> Moreover, in a 2.1 device I saw a lack of connection for a couple of minutes (where many packages were lost). I need more time to see if it's an isolated event.

I set up other 2.1 devices and I have this issue on all of them: they're losing more "pings" than 2.0 ones. Do you prefer closing this issue and creating a new one (maybe with a better title)?

marcelstoer, 2017-05-25:
> Do you prefer closing this issue and creating a new one

It's definitely helpful if you don't have to go through lots of comments to arrive at a conclusion.

How, if at all, is this related to #1973?

devsaurus, 2017-05-25:
Yes, new issue for different topic as Marcel stated.

Regarding this one I'd like to understand better what's going on.
I can't reproduce that a lease renewal resulting in the same IP actually takes down the interface including active connections. Specifically, the `wifi.eventmon.STA_GOT_IP` event is *not* triggered for me when a renewal for same IP happens.
The lwIP code contains a respective check at [L329](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/lwip/core/netif.c#L329) and debug logs from dhcp and netif suggest that it's working.

Aterocana, 2017-05-26:
So maybe it has nothing to do with DHCP lease? Do your devices stay "up" for more than 2 hours without "connection to broker failed"?

closed 2017-05-26 by Aterocana