nodemcu/nodemcu-firmware #1988: nodemcu-build.com: cjson & sjson

Code1110, 2017-05-24:
### Actual behavior
In 2.1.0-master_20170521 cjson has been replaced by sjson. On nodemcu-build.com I can only choose to include the cjson module, which is then not available in the builded firmware. cjson should be replaced by sjson.

### Test code
```Lua
> =cjson
nil
> =sjson
nil
> =file
romtable: 0x40263900
> =gpio
romtable: 0x40263d20
> =http
romtable: 0x40263e98
> =net
romtable: 0x402643a0

```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e
	SSL: false
	modules: cjson,file,gpio,http,net,node,tmr,uart,wifi
 build 	built on: 2017-05-24 22:44
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
WeMos D1 mini

Aterocana, 2017-05-25:
I have no problem with sjson:
```
=sjson
romtable: 0x402a2e20
```
I guess the issue is with nodemcu-build.com since I've compiled mine manually.

marcelstoer, 2017-05-25:
Thanks for reporting, it's fixed.

```
=cjson
nil
> =sjson
romtable: 402960e4
```

```
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e
	SSL: true
	modules: cron,dht,enduser_setup,file,gpio,http,net,node,rtctime,sjson,sntp,spi,tmr,uart,wifi,tls
 build 	built on: 2017-05-25 11:19
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
```

closed 2017-05-25 by marcelstoer