nodemcu/nodemcu-firmware #1989: sjson should trim trailing null bytes

rtahiry, 2017-05-25:
### Expected behavior
sjson.decode() (and possibly encode()) should trim trailing null bytes. This is particularly important when the json string was encrypted and usually terminated by a null byte. 
### Actual behavior
sjson.decode() throws a FOUND_NULL_BYTE exception when string has trailing null byte. The following snippet shows this behavours and confirms that everything goes back to normal when trailing null bytes are removed.
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
=sjson.decode('{}')
table: 3fff0408
> cypher=crypto.encrypt("AES-ECB","1234567891234567", '{}')
> text=crypto.decrypt("AES-ECB","1234567891234567", cypher)
> =text
{}
> =sjson.decode(text)
stdin:1: JSON parse error: FOUND_NULL_BYTE
stack traceback:
	[C]: in function 'decode'
	stdin:1: in main chunk
> text=text:gsub('%z.*','')
> =sjson.decode(text)
table: 3fff0408
> 
```
### NodeMCU version
Nodemcu dev branch built from nodemcu-build.com with sjson and crypto library. (master branch as a broken cjson library #1988)

adoconnection, 2017-06-05:
+1 because its breaking change.
sjson.decode is not compatible with cjson.decode because of this

pjsg, 2017-06-05:
I can certainly change the default behavior so that it ignores data after the end of the decoded object. I think that the padding done by the crypto.encrypt is probably wrong in this case (padding with nulls is not recommended). The most commonly used padding is to fill out the block with repeated bytes indicating how many bytes are pads. See RFC 5652, RFC 1423.

adoconnection, 2017-06-05:
> the padding done by the crypto.encrypt is probably wrong 

You are absolutely right. Yet, old implementation in cjson.decode handled this case correcly.

marcelstoer, 2017-06-05:
> Yet, old implementation in cjson.decode handled this case correcly.

It's never too late to correct old issues...

PR #1999 (2017-06-18): Only handle errors if we have not parsed a complete object
Fixes #1989 and #2005 .

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This ignores errors that happen in the parser after the object is parsed. This make `sjson` behave like `cjson`.
closed 2017-06-18 by marcelstoer