nodemcu/nodemcu-firmware #1991: wifi functions not working properly on ESP-01

krjj, 2017-05-27:
Issues : 
1) Cannot connect to the wifi station
2) Cannot setup access point
3) Cannot get list of stations

### Actual behavior
wifi.sta.status() return 3 (Wrong password) code when trying to connect to station

### NodeMCU version
2.1.0 (master)

### Hardware
ESP-01 Module

I did not encounter any problems with previous version (2.0.0).

dnc40085, 2017-05-29:
Hello, 
I just tested both the `master` and `dev` branches on my ESP-01 and all of the wifi functions seem to be working fine.

When you updated from 2.0.0 to 2.1.0 did you erase the flash?  
It is possible for the configuration sectors to be incompatible between SDK versions 2.0.0 and 2.1.0.

krjj, 2017-05-29:
Erasing the flash memory solved the issue

closed 2017-05-29 by krjj