nodemcu/nodemcu-firmware #2000: DHCP Lease shuts down connection even if it's just renewed

Aterocana, 2017-06-08:
8<------------------------ BUG REPORT -----------------------------------------
I have some esp8266 which lose connection for a while when the DHCP lease expires, even if they get the same IP as before. This does not happen on nodeMCU-1.5.4.1.
### Expected behavior
The esp8266 stays connected.

### Actual behavior
The esp8266 has some seconds where it's not connected to AP.

### Test code
```Lua
wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, function(T)
	print("\n\tSTA - CONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
	T.BSSID.."\n\tChannel: "..T.channel)
end)

wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T)
	print("\n\tSTA - DISCONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
	T.BSSID.."\n\treason: "..T.reason)
end)

wifi.eventmon.register(wifi.eventmon.STA_AUTHMODE_CHANGE, function(T)
	print("\n\tSTA - AUTHMODE CHANGE".."\n\told_auth_mode: "..
	T.old_auth_mode.."\n\tnew_auth_mode: "..T.new_auth_mode)
end)

wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, function(T)
	print("\n\tSTA - GOT IP".."\n\tStation IP: "..T.IP.."\n\tSubnet mask: "..
	T.netmask.."\n\tGateway IP: "..T.gateway)
end)

wifi.eventmon.register(wifi.eventmon.STA_DHCP_TIMEOUT, function()
	print("\n\tSTA - DHCP TIMEOUT")
end)

wifi.eventmon.register(wifi.eventmon.AP_STACONNECTED, function(T)
	print("\n\tAP - STATION CONNECTED".."\n\tMAC: "..T.MAC.."\n\tAID: "..T.AID)
end)

wifi.eventmon.register(wifi.eventmon.AP_STADISCONNECTED, function(T)
	print("\n\tAP - STATION DISCONNECTED".."\n\tMAC: "..T.MAC.."\n\tAID: "..T.AID)
end)

wifi.eventmon.register(wifi.eventmon.AP_PROBEREQRECVED, function(T)
	print("\n\tAP - PROBE REQUEST RECEIVED".."\n\tMAC: ".. T.MAC.."\n\tRSSI: "..T.RSSI)
end)

wifi.eventmon.register(wifi.eventmon.WIFI_MODE_CHANGED, function(T)
	print("\n\tSTA - WIFI MODE CHANGED".."\n\told_mode: "..
	T.old_mode.."\n\tnew_mode: "..T.new_mode)
end)
```

Every time my DHCP expires (every 2 hours) I get this:
```
STA - GOT IP
Station IP: 192.168.0.164
Subnet Mask: 255.255.255.0
Gateway IP: 192.168.0.1
```
Even if my IP is the same as previous connections.
### NodeMCU version
I'm on 2.1. The problem arises also on 2.0, while on 1.5.4.1 doesn't happen.

### Hardware
Many evaluation board types: nodemcu AI Thinker, nodemcu LOLIN, wroom02...

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2017-06-09:
I don't see anywhere that your device is actually "loosing connection", I only see (imply) that the `STA_GOT_IP` event is fired with every DHCP lease renewal. May be related to #1936. Did you test that the connection is dropped e.g. by pinging every 500ms during the lease renewal period?

Aterocana, 2017-06-10:
Sorry about that, I forgot to mention I have a GET request every second which works always but when I get my DHCP session renewed. I had the same issue here: [#1984](https://github.com/nodemcu/nodemcu-firmware/issues/1984).

closed 2019-06-24 by stale[bot]