nodemcu/nodemcu-firmware #2007: HTTP.get no callback on error

HHHartmann, 2017-06-20:
http.get on a local IP that is not available (not assigned) fails but does not call callback

### Expected behavior
The callback should be called  **allways**, even on all failures

in the example it should print
callback

### Actual behavior
The callback is not called

-- > http.get("http://192.1.1.254/",nil,function() print "callback" end)
-- > HTTP client: Disconnected with error: -11
-- HTTP client: Connection timeout
-- HTTP client: Connection timeout

### Test code

```Lua
-- http.get("http://192.1.1.254/",nil,function() print "callback" end)
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e
	SSL: true
	modules: bit,file,gpio,http,i2c,net,node,ow,pwm,rtctime,sjson,sntp,tmr,uart,wifi,tls
 build 	built on: 2017-06-17 17:53
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
Standard ESP6288 E-12 Dev Board

wnasich, 2017-06-21:
I have same behavior with http.post()

marcelstoer, 2017-06-21:
I'd argue the current behavior is correct. The callback parameter is specified/documented as

> The callback function to be invoked when the response has been received; it is invoked with the arguments status_code, body and headers

In your case no connection can be established and thus no status code, no body and no headers.

HHHartmann, 2017-06-21:
yes but the doc also says
>Each request method takes a callback which is invoked when the response has been received from the server. The first argument is the status code, which is either a regular HTTP status code, or -1 to denote a DNS, connection or out-of-memory failure, or a timeout (currently at 10 seconds).

marcelstoer, 2017-06-21:
Interesting, never noted that paragraph.

HHHartmann, 2017-06-21:
It would also be useful (at least) to know when processing of an http call is finished, so I can issue the next call.

vowstar, 2017-06-22:
Yes, you are right. When timeout it need a callback with error code -1.

marcelstoer, 2017-07-01:
PR merged -> closing

PR #2020 (2017-07-01): fix #2007 http: Call callback in all cases
Call callback with errorcode -1 if no connection could be establioshed

Fixes #2007 and #1721.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected i [... cut]
closed 2017-07-01 by marcelstoer