nodemcu/nodemcu-firmware #2010: wifi (doc) questions

danjohn, 2017-06-22:
Hi/,
please add to docs (+code, maybe):

* wifi.sta.clearconfig() immediately breaks an existing connection (odd, if you just want to write a new config to flash and keep online status)
* after that, wifi.sta.get[default]config() returns old values (should be nil)
* wifi.sta.getapindex() returns 1 (should be nil)

It's not clear from the docs, how to juggle with wifi configs: How to decide when to clear/recover to a default config (virgin/crashed device), in which slot does a config(save=true) land, how to rewrite a specific slot, does some algo automagically select the best config (signal strength), does it switch networks when the device is moving?

I suppose, the SDK imposes that shortcomings, sorry for couriosity.
TY

danjohn, 2017-06-26:
I did some empirics:

    >wifi.sta.config({ssid="xx", pwd="xx", save=true})
writes to Slot 1, and connects to xx
    >wifi.sta.config({ssid="yy", pwd="xx", save=true})
writes to Slot 1, disconnects immediately (not documented)
    >wifi.sta.changeap(2) = **false**
obviousely there is only one slot (doc says: default is 5)
    >wifi.setaplimit(3) = **true**
    >wifi.sta.changeap(2) = **false**
obviousely there is still only one slot?
    >wifi.setaplimit(2) = **true**
    >wifi.sta.changeap(2) = **false**
    >wifi.sta.config({ssid="yy", pwd="xx", save=true})
writes to Slot 1
obviousely only one slot, again

I don't know, which ESP the **readthedos.io** doc is based on, but it doesn't match reality for me (`setaplimit()` pretends to *do* (returns **true**), but *doesn't*).
Have I detected a NodeMCU issue, or is this just a documentation flaw? ;-)

(Hardware: AI-THINKER ESP8266MOD, 
NodeMCU comit: c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e)

dnc40085, 2017-06-27:
>Have I detected a NodeMCU issue, or is this just a documentation flaw? ;-)

With the function `wifi.sta.setaplimit()` it is a documentation issue that has been partially remedied on the dev branch.
I updated the `dev` branch documentation with the correct default value (`1`), but I forgot to add that in order for the new `wifi.sta.setaplimit()` value to take effect NodeMCU must be restarted.

As for `wifi.sta.clearconfig()` breaking an existing connection, this is due to the function using the sdk function `wifi_station_set_config()` to clear the config by overwriting the current configuration.

Concerning your other questions:
 * `How to decide when to clear/recover to a default config (virgin/crashed device)?` the Espressif documentation gives no details about this, the only info I've been able to find about the restore API (exposed by `node.restore()`) is what other APIs it restores to default.

 * `in which slot does a config(save=true) land?` I believe this is random.

 * `how to rewrite a specific slot?`  This feature does not (currently) exist, however I have been working on an API that would allow this along with giving the option to update the stored wifi config as a whole via a Lua table

 * `does some algo automagically select the best config (signal strength)?` `does it switch networks when the device is moving?` The Espressif documentation does not mention either of these features, so I'm not sure.

marcelstoer, 2017-06-27:
> I forgot to add that in order for the new `wifi.sta.setaplimit()` value to take effect NodeMCU must be restarted

Happy to merge a PR whenever you feel like it 😄 

I'll close this as the rest are questions. Thanks for answering!

PR #2017 (2017-06-28): Update documentation for wifi.sta.setaplimit()
Fixes #2010

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/en/*`.

This PR adds a note to the `wifi.sta.setaplimit()` documentation that lets the developer know that the new setting will not take effect until restart.
closed 2017-06-27 by marcelstoer