nodemcu/nodemcu-firmware #2015: Remove Hardware FAQ from the NodeNCU documentation

TerryE, 2017-06-26:
### Feature retirement

Remove Hardware FAQ from the NodeMCU documentation

### Justification

I've been making lots of notes in a potential FAQ content format about a bunch of H/W issues related to putting NodeMCU Lua to work for some of my real applications.  This issue actually started life as a discussion about including this in our documentation set.  The issue _as raised above_ came out of my asking the Q: does would issues and changes relating to this content belong on this list?  In my mind the answer is a firm _no_.  This scope of this project is about developing and maintain a Lua based firmware / RTS for the ESP chipsets.  It is not about ESP hardware related issues.  Places like esp8266.com are where to go for this type of dialogue.  So I have decided to add my content to my wiki pages there.

However, this decision also begs the question about our current H/W FAQ topics.  They are fragmentary and out dated and a clone of content on esp8266.com.  My proposal is that the don't belong in our documentation set and should be removed.  
  
### Workarounds

N/A as we can leave the current content as-is and let it continue to get more dated and irrelevant.

TerryE, 2017-06-27:
Mark I've updated the pages on esp8266.com; I leave you to do the corresponding changes on our doc set :smile: Terry

marcelstoer, 2017-06-27:
I guess this was directed at me...fixed.

commit 583afc0f (2017-06-27): Remove hardware FAQ, fixes #2015
closed 2017-06-27 by marcelstoer