nodemcu/nodemcu-firmware #2016: Error in unpacking a float

kc2h, 2017-06-26:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
That the unpack would import the float correctly.

### Actual behavior
The import is adding random values to the data.
Errors occur with ALL imported floats.
16 bit int (I2) imports are working correctly.

### Test code
-- Read structure from arduino via i2c.
    i2c.start(id)
    i2c.address(id, dev_addr, i2c.RECEIVER)
    c = i2c.read(id, 28)
    i2c.stop(id) 

-- Debug the incoming data
    local debugFloat = ""
    for i=1,4 do
        debugFloat = debugFloat .. string.format("%02x",string.byte(c,i))
    end
    print("Left Float = " .. debugFloat)
-- Sample data is :  Left Float = 66e6f642
-- 0x66e6f642 converted from float is: 123.45

    local idx = 1;
    i2cData["Left"]["Balance"], idx = struct.unpack("f", c, idx)
    print(string.format("Left.Balance: %f", i2cData["Left"]["Balance"]))
-- Prints:  Left.Balance: 123.449997

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e
	SSL: true
	modules: crypto,enduser_setup,file,gpio,http,i2c,mqtt,net,node,sjson,struct,tmr,uart,wifi,tls
 build 	built on: 2017-06-25 21:26
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
ESP8266MOD ESP-12

8<------------------------ END BUG REPORT -------------------------------------

pjsg, 2017-10-26:
It turns out that 123.45 cannot be represented exactly in IEEE-754 floating point. If you go to https://www.h-schmidt.net/FloatConverter/IEEE754.html you can enter 123.45 and see the result. in short

```
0x42f6e666 -> 123.4499969482421875
0x42f6e667 -> 123.45000457763671875
```

I.e. 123.45 is roughly halfway between the two closest values that are representable as floating point numbers.

pjsg, 2017-10-29:
If you use "%.2f" in your format string, then it would display the number that you are expecting.

closed 2017-10-29 by pjsg