nodemcu/nodemcu-firmware #2023: sjson module causes unresolved symbols

jmattsson, 2017-07-03:
### Expected behavior
 - No unresolved symbols after linking the firmware ELF.
 - Firmware build should fail if there are unresolved symbols.

Initially observed by @dnc40085 over in #1755. Historically `esptool.py` has bailed out if there are undefined symbols, but clearly the current version we have in our tree does not.

I have no idea why this wasn't failing horribly for @pjsg during testing...

### Actual behavior
```
../tools/esptool.py elf2image --flash_mode dio --flash_freq 40m .output/eagle/debug/image/eagle.app.v6.out -o ../bin/
esptool.py v1.2-dev
Warning: ELF binary has undefined symbol calloc
Warning: ELF binary has undefined symbol free
```
### Test code
Include the SJSON module in the build.

### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
`dev`

### Hardware
n/a

jmattsson, 2017-07-03:
So it seems none of the functions which use `calloc`/`free` aren't actually used by the Lua module, and thus haven't caused actual problems. Now to work out why those functions are even included in the ELF after the linking...

jmattsson, 2017-07-03:
Erk, why did we end up `#include`ing a .c file here?

jmattsson, 2017-07-03:
Okay, I'm getting an appreciation for why a harsh `#include` was picked... curse the NON-OS SDK's non-standard include files to Hel and back **sigh**

jmattsson, 2017-07-03:
Fascinating. If I build jsonsl.c into a library like everything else, things break at runtime.

closed 2017-07-04 by jmattsson

gbox3d, 2017-09-08:
resut = sjson.encode({code="print(\"hello\")"})
result is
"{\"code\":\"print(\"hello\")\"}"
it may be wrong

right result is "{\"code\":\"print(\\\"hello\\\")\"}" 

it is bug?

PR #2024 (2017-07-04): Sjson undefined symbol fix
Fixes #2023 .

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

This PR fixes the `undefined symbol` warnings that appear when including the `sjson` module.
it also adds a check to catch any future such warnings and bail out the build in such case.