nodemcu/nodemcu-firmware #2025: Is it valid for wifi.sta.getapinfo() to return a null BSSID?

marcelstoer, 2017-07-03:
Running the example from https://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#wifistagetapinfo I see that I get `null` for my AP's BSSID. Is that an ok condition or does that signify a bug in the SDK or our mapping to Lua?

Maybe something for @dnc40085?

dnc40085, 2017-07-03:
Yes, that is an ok condition.
From the [wifi.sta.getapinfo()](https://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#returns_19) documentation
>`bssid` MAC address of Access Point, `nil` if no MAC address was configured

Unless your station was configured with the AP's mac address( ex: `wifi.sta.config({ssid=ssid, pwd=pwd, bssid="AA:BB:CC:DD:EE:FF"})`), the `bssid` field should be `nil`.

The reasoning behind this is that when no BSSID is set during station configuration, the `station_config.bssid_set` byte is set to `0` and the 6 byte `station_config.bssid` array is ignored and the corresponding location in flash is not modified. This location being unmodified could potentially contain a BSSID that does not correspond with the SSID so it must be ignored when the access point info is retrieved from flash.

Considering the confusion, maybe it would be prudent to change it to return an invalid mac address(`00:00:00:00:00:00`) instead of `nil`.

marcelstoer, 2017-07-03:
Thanks for the explanation.
Before I posted this question I checked the documentation but "`nil` if no MAC address was configured" confused me. Why would I configure the AP's MAC I thought, it's got a MAC automatically. It didn't occur to me that this would be the MAC I set explicitly while configuring the station.
Then I looked at https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/wifi.c#L564 and tracked the use of `bssid_set`. However, not being a firmware developer this didn't fully resolve my confusion. Hence, this issue.

Conclusion: there's no way we could figure out the AP's MAC from Lua land unless we knew it beforehand?

dnc40085, 2017-07-04:
>Conclusion: there's no way we could figure out the AP's MAC from Lua land unless we knew it beforehand?

The submitted PR should give you a way to get the currently configured AP's MAC address.

PR #2026 (2017-07-08): Modify wifi.sta.get*config() to return AP's MAC
Fixes #2025.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/en/*`.

This PR modifies the functions `wifi.sta.getconfig()` and `wifi.sta.getdefaultconfig()` to output the BSSID even if station was originally not configured to connect to an AP  [... cut]
commit 2e33abe1 (2017-07-08): Modify wifi.sta.get*config() to return AP's MAC (#2026)

* Modified wifi.sta.get*config() to return AP's MAC even if bssid_set==0
* Improved documentation for wifi.sta.getapinfo, fixes #2025
closed 2017-07-08 by marcelstoer